dji-sdk/Onboard-SDK-ROS #103: Corrupted output from /dji_sdk/gimbal

asiron, 2017-07-06:
The output of `/dji_sdk/gimbal` topic is corrupted as the Onboard SDK OPEN protocol describes :

**"Note: Rotating 90 degree in pitch direction will cause gimbal lock problem, in which the value of roll and yaw are not reliable."**

This makes us unable to fly and point the camera downwards as the value of yaw and roll is not reliable. It is very important for us to always have a reliable reading from the gimbal in order to transform into the camera frame. Furthermore, once we pass the 90 degree value, it does not continue to 180 degrees, but starts decreasing to 0 degrees again.

Are you planning on fixing it?

asiron, 2017-07-06:
I actually haven't check if the same issue appears in the Zenmuse X3 gimbal. I will check it tomorrow and update the issue

asiron, 2017-07-07:
This issue does not appear with the Zenmuse X3 gimbal

zliDJI, 2017-07-10:
Hi, asiron, thanks for your feedback and your effort to test over multiple gimbals. We will test the issue on our side later.

closed 2021-06-25 by dji-kyle