dji-sdk/Onboard-SDK-ROS #106: local_position gone in 3.3?

damanfb, 2017-07-11:
Is it correct that local_position is gone in the 3.3 sdk? Before, the dji assistant 2 simulator would only update local_position after flying around. The gps coordinates would maintain a constant value at the initial seed. If this is still true, than I cannot use the simulator if the local_position is gone and that issue is not fixed.

rohitsantdji, 2017-07-12:
local_position is not available in OSDK 3.3 yet, but the simulated GPS coordinates are correctly updated (even though the precision of the Assistant 2 telemetry display is not enough to show changes in GPS coordinates for small changes). You can try out the position control sample code which uses simulated GPS coordinates (and the localOffsetFromGPSOffset function) for closed-loop control.

closed 2017-07-12 by rohitsantdji

amenonDJI, 2017-09-11:
@damanfb Local Position Topic was released on Friday September 8th along with a Local position control topic. The new Local position support allows the user to set the local position reference/origin wherever he/she chooses to with a service call. Give it a try.