dji-sdk/Onboard-SDK-ROS #108: SDK authority, control and other small problems

chutsu, 2017-07-14:
So far I'm not very impressed with the latest Onboard-SDK version, the new update is plagued with drastic backward compatibility issues and the behaviour is completely different to what it was before, worst of all the SDK does not seem stable at all.

Problems I have:

1. Acquiring / Releasing SDK Authorization is not robust, here is a screen shot where I call the ROS Service `/dji_sdk/sdk_control_authority` where I can obtain SDK control of my N3 but your SDK code seems to spit out `UNKNOWN_ACK_ERROR_CODE`, should I be concerned? Further, I cannot release SDK control reliably, I know this because I inspect the `dji_sdk/SDKControlAuthority` response variable `result` to check if I have indeed released SDK control, you can see from the screen shot below that the last line shows where it failed to release.

I don't think I need to explain why this is important. 

[image]

2. I have a specific use case where I would like to control the quadrotor's attitude, this means I want to control: Roll, Pitch, Yaw and Thrust (0 - 100). This should correspond to the DJI Control Flag Byte of `0x20`

```
    //  DJI Control Flag Byte
    //
    //    bit 7:6  0b00: HORI_ATTI_TILT_ANG
    //             0b01: HORI_VEL
    //             0b10: HORI_POS
    //
    //    bit 5:4  0b00: VERT_VEL
    //             0b01: VERT_POS
    //             0b10: VERT_THRUST
    //
    //    bit 3    0b0: YAW_ANG
    //             0b1: YAW_RATE
    //
    //    bit 2:1  0b00: horizontal frame is ground frame
    //             0b01: horizontal frame is body frame
    //
    //    bit 0    0b0: non-stable mode
    //             0b1: stable mode
    //
    //  We used:
    //
    //    HORIZ_ATTI_TILT_ANG
    //    VERT_THRUST
    //    YAW_ANG
    //    ground frame
    //    non-stable mode
    //
    //  ends up being: 0b00100000 -> 0x20
```
If I am correct, I want my roll, pitch and yaw to be respective to the ground frame, and my thrust indeed should be in the body frame, so why am I getting the ROS warning as per screen shot above?

3. In the previous SDK version your API expected the control angles to be in degrees, you have put an attention warning at the start of the ROS wiki but I would argue this is not emphasised well enough, I just have to infer from the docs you wanted radians for angles. This is a small nuance, but for developers who are migrating from previous version this is an important thing to note.

4. Removal of `local position` ROS topic, I'm not sure what this topic was removed. For research purposes this is still a useful topic to have, I want able to control with respect to a defined home point. This is particularly useful if researchers want to use the DJI flightstack indoors, does DJI plan on supporting MOCAP/Vicon support in the near future?

rohitsantdji, 2017-07-15:
Here are some comments regarding your issues:

1. We came across this bug and fixed it internally, you should see a fix in the next release.
2. This kind of warning was put in place to make it explicit that when you set the coordinate frame, some elements of your control inputs will still be in certain pre-set frames of reference. The notion of using a "horizontal" frame that developers can set, while having vertical frames pre-set is somewhat unusual and we decided to emit warnings for cases when the x-y (and yaw) inputs are in a different frame than the z-input. In your case, you can safely ignore the warning. We'll continue to collect feedback about this design and potentially remove it in the future.
3. Thanks for the feedback - we mention compatibility with REP-103 which defines radians as a standard unit for angles, but we'll make it clearer.
4. `local_position` is on our roadmap and will be in a forthcoming release! 

We really appreciate the detailed feedback! The large 3.3 refactor is intended to bring the SDK in line with developer expectations for a robotics framework and ROS best practices, even as we continue to add more features. We realize that this release requires a large change in the way developers use the SDK, especially in ROS, and also that the SDK is a little rough around the edges this time around. I'd say these are side effects of a ground-up refactor like this one, and we'll continue to squash bugs and improve some design elements in the coming releases.

amenonDJI, 2017-08-16:
Closing this issue. Please raise a new issue if there are any other questions.

closed 2017-08-16 by amenonDJI

amenonDJI, 2017-09-11:
@chutsu #4 - Local position topic support was released on Friday September 8th. Give it a try.