dji-sdk/Onboard-SDK-ROS #110: Python support question

jswoboda, 2017-07-21:
Are you still supporting the python wrapper after v 3.2? You still have the setup.py script in there but drone.py is gone. Can I just drop drone.py from 3.2?

zliDJI, 2017-07-24:
Hi, @jswoboda . The Onboard-SDK-ROS itself is already a wrapper, which publishes/subscribes topics, and provided services.  Client codes just subscribe to the topics and call the services, and codes can be written in either C++ or python. 

At this moment we don't see necessity of creating another wrapper, since all it does it to subscribed all topics and services. If you have special needs for a python wrapper, we are happy to know.

rohitsantdji, 2017-08-09:
Closing due to no follow up from OP. If there's still lingering questions, please re-open. Thanks!

closed 2017-08-09 by rohitsantdji

jswoboda, 2017-09-14:
Hi, thanks for responding. I actually have had great success with your Python wrapper from ROS OSDK 3.2. It helped speed up my development and testing quite a bit. I understand after working with it that's it's just subscribing and calling services but this was my first time using ROS so it this was new for me.

There's a large interest in drones within the geosciences community for doing various kinds of measurements. Our community uses a lot of Python and having a wrapper can cut down on development time so we can focus on our science.  

Sorry for the late reply. I'm currently sticking with v 3.2 for now. I'm using a Matrice 600 and I haven't updated the firmware to use v 3.3 with it. Can I get away with not doing this update? If I need to update the firmware can I still use drone.py?