dji-sdk/Onboard-SDK-ROS #112: Could not find a package configuration file provided by "geometry_msgs"

yihuanliao, 2017-07-26:
Hi, I am new to this developing things. I am following the guide documentations. But I have met some errors when proceeding to "catkin_make"

Here is the error which I encountered:

CMake Error at /usr/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
Could not find a package configuration file provided by "geometry_msgs"

Can anyone please help me?
Many thanks

zliDJI, 2017-07-26:
Can you give the detailed procedure how you installed ROS? Did you follow this link? 
http://wiki.ros.org/kinetic/Installation/Ubuntu

Most users install ROS under /opt/ros instead of /usr. Also, install  ros-kinetic-desktop-full if possible.

yihuanliao, 2017-08-01:
Yes, I have done so, but I still have this problem, seems like there are some files missing.
[image]
Can anyone please help me with that? (I know this feels like a stupid question, but I am really new to SDK development)
Many thanks

zliDJI, 2017-08-03:
Please check if this link can help you.
https://github.com/mavlink/mavros/issues/539

amenonDJI, 2017-08-16:
@yihuanliao Closed due to inactivity. Please reopen if you continue to have issues.

closed 2017-08-16 by amenonDJI

AbhishekGS, 2019-08-07:
Hi @yihuanliao , @zliDJI , @amenonDJI,

I am facing the same issue while trying to execute the command catkin_make on Raspberry Pi , which is running Raspbian. 

[image]

I followed the instructions from the below link for installing ROS kinetic on Raspberry Pi.
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

I tried executing the below command as mentioned in [mavlink/mavros#539],(https://github.com/mavlink/mavros/issues/539) 

`sudo apt-get install ros-kinetic-geometry-msgs`

However, I get the below error.

[image]

Please guide me.

Thanks,
Abhishek

yihuanliao, 2019-08-07:
Try

sudo apt install ros-kinetic-geometry-msgs

This might work.

It has been a long time, I couldn’t remember whether I solved this problem.

Best regards,
Yihuan

TianleZhang-sketch, 2021-01-05:
I met the same error,i've  tried  to download the geometry package,but it didn't work.So i watch carefully about the error  reminder，i find it is because i  made a spell mistake when i use"catkin_create_pkg  test_pkg  depend1 depend2"to create a ros package.So my suggestion is to check the find_pkg from your CmakeList.txt

DanGlChris, 2022-05-15:
I met this problem when I tried to install [**rosserial** ](https://atadiat.com/en/e-rosserial-arduino-introduction/#:~:text=Clone%20the%20source%20code%20in%20this%20directory) package on Raspbian (buster)
I tried 
- **sudo install ros-kinetic-desktop-full** but It doesn't work
- **Also sudo apt-get install ros-kinetic-geometry-msgs** the same problem **Package not found**
- then I tried **ros-geometry-msgs**. package was installed but when I tried to reinstall **rosserial**  geometry-msgs was again not founded

so I decide to install it manually
I clone this [**repository** ](https://github.com/ros/common_msgs) inside the src on my catkin's workspace

I restart the building (with catkin_make)
and restart the installation of rosserial
IT WORK!!!

Chen001117, 2022-08-12:
> https://github.com/ros/common_msgs
> https://answers.ros.org/question/303858/errors-while-using-catkin_make-in-ros-inside-raspbian-stretch/
these sites solve my problem.