dji-sdk/Onboard-SDK-ROS #115: SDKInfo feedback disappeared in 3.3

edkoch, 2017-08-04:
Maybe I'm missing something, but I noticed that device status (i.e. SDKInfo) disappeared in 3.3. It is a valuable piece of feedback that is use to determine who has control of the FCU. I went ahead and added a simple std::Uint8 topic to dji_sdk_node_publisher.cpp in the 50Hz section and confirmed that it works as expected. See the code snippet below. My question is whether you guys intend to add back in the whole SDKInfo feedback in a future release or should I go ahead and submit a pull request for what I did. Note that for my purposes I am only feeding back the deviceStatus and not the other fields in SDKInfo.

  //update device control info
  if (data_enable_flag & DataBroadcast::DATA_ENABLE_FLAG::A3_HAS_DEVICE)
  {
    std_msgs::UInt8 status_device;
    status_device.data = vehicle->broadcast->getSDKInfo().deviceStatus;
    p->device_status_publisher.publish(status_device);
  }

edkoch, 2017-08-04:
BTW given that there is bug in the return value of the SDK Authority service it is impossible to know if you got authority or not without this feedback. I have been able to confirm that even though the SDK Authority service returns false the SDK still gets authority as evidenced by the deviceStatus being set to 2.

rohitsantdji, 2017-08-04:
Hi @edkoch ! In the 3.3.1 release, we've fixed the control authority service bug. Could you give it a shot?

amenonDJI, 2017-08-16:
@edkoch Have you had a chance to test this out?

edkoch, 2017-08-16:
Yes I did and it seems to work.

thanks,

-ed koch

amenonDJI, 2017-08-16:
Great thanks. Closing this issue.

closed 2017-08-16 by amenonDJI