dji-sdk/Onboard-SDK-ROS #116: SDK compiling issue with RPi3 (Ubuntu Mate 16.04 with ROS Kinetic)

rocklinsuv, 2017-08-05:
Hi there,

Recently I've experienced an issue when compiling the Onboard-SDK-ROS package. I'm currently using a Raspberry Pi 3 with Ubuntu Mate 16.04 and ROS Kinetic installed. First I downloaded Onboard-SDK and compiled it without any issue. However, when I tried to catkin_make the Onboard-SDK-ROS package, it stuck at 90% every time and the system froze. Here are some screen shoots I've captured:

I tried Onboard-SDK v3.2 with Onboard-SDK-ROS v3.2:
[image]

Onboard-SDK v3.3 with Onboard-SDK-ROS v3.3
[image] 

Even the newest Onboard-SDK v3.3.1 with Onboard-SDK-ROS v3.3.1
[image] 

It seems like the dji-sdk-node would not compile. I previously tried to compile everything in a Virtual Machine with Ubuntu 16.04 and ROS Kinetic without any issue. Any thoughts that may help me resolve this problem? Many thanks!

Cheers,

andre-nguyen, 2017-08-06:
On memory limited systems you can try to compile with only one job. e.g.
```
catkin_make -j1
```

rocklinsuv, 2017-08-06:
Woaw! It worked like a charm! Many thanks mate :smiley:

closed 2017-08-09 by rohitsantdji