dji-sdk/Onboard-SDK-ROS #118: onboard-sdk and onboard-sdk-ros

mithunvarma, 2017-08-15:
cannot catkin build or catkin_make the new sdk 3.3.1 version.

even the cmakelist of guidance packages have the errors like the project name is dji_sdk instead of guidance-sdk

can any one help me fix the following issue
Thanks in advance

Errors     << dji_sdk:make /home/user/directory/logs/dji_sdk/build.make.000.log
/home/user/directory/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionWpGetInfoCallback(dji_sdk::MissionWpGetInfo::Request&, dji_sdk::MissionWpGetInfo::Response&)’:
/home/user/directory/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:249:48: error: ‘class DJI::OSDK::WaypointMission’ has no member named ‘getWaypointSettings’
     info = vehicle->missionManager->wpMission->getWaypointSettings(10).data;
                                                ^
make[2]: *** [CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_mission_services.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/dji_sdk_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/user/directory/build/dji_sdk; catkin build --get-env dji_sdk | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

mithunvarma, 2017-08-15:
tried to fix the above error, then i got the following error for dji_sdk_node_publisher.cpp

CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_publisher.cpp.o: In function `DJISDKNode::dataBroadcastCallback()':
dji_sdk_node_publisher.cpp:(.text+0xfc): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x169): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x1d6): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x243): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x2b0): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_publisher.cpp.o:dji_sdk_node_publisher.cpp:(.text+0x309): more undefined references to `DJI::OSDK::DataBroadcast::getRC() const' follow
CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_publisher.cpp.o: In function `DJISDKNode::dataBroadcastCallback()':
dji_sdk_node_publisher.cpp:(.text+0x3c1): undefined reference to `DJI::OSDK::DataBroadcast::getQuaternion() const'
dji_sdk_node_publisher.cpp:(.text+0x414): undefined reference to `DJI::OSDK::DataBroadcast::getQuaternion() const'
dji_sdk_node_publisher.cpp:(.text+0x46e): undefined reference to `DJI::OSDK::DataBroadcast::getQuaternion() const'
dji_sdk_node_publisher.cpp:(.text+0x4c4): undefined reference to `DJI::OSDK::DataBroadcast::getQuaternion() const'
dji_sdk_node_publisher.cpp:(.text+0x718): undefined reference to `DJI::OSDK::DataBroadcast::getAcceleration() const'
dji_sdk_node_publisher.cpp:(.text+0x77e): undefined reference to `DJI::OSDK::DataBroadcast::getAcceleration() const'
dji_sdk_node_publisher.cpp:(.text+0x7ef): undefined reference to `DJI::OSDK::DataBroadcast::getAcceleration() const'
dji_sdk_node_publisher.cpp:(.text+0x860): undefined reference to `DJI::OSDK::DataBroadcast::getAngularRate() const'
dji_sdk_node_publisher.cpp:(.text+0x8b6): undefined reference to `DJI::OSDK::DataBroadcast::getAngularRate() const'
dji_sdk_node_publisher.cpp:(.text+0x917): undefined reference to `DJI::OSDK::DataBroadcast::getAngularRate() const'
dji_sdk_node_publisher.cpp:(.text+0xa27): undefined reference to `DJI::OSDK::DataBroadcast::getGlobalPosition() const'
dji_sdk_node_publisher.cpp:(.text+0xbc5): undefined reference to `DJI::OSDK::DataBroadcast::getVelocity() const'
dji_sdk_node_publisher.cpp:(.text+0xc1b): undefined reference to `DJI::OSDK::DataBroadcast::getVelocity() const'
dji_sdk_node_publisher.cpp:(.text+0xc71): undefined reference to `DJI::OSDK::DataBroadcast::getVelocity() const'
dji_sdk_node_publisher.cpp:(.text+0xd51): undefined reference to `DJI::OSDK::DataBroadcast::getBatteryInfo() const'
dji_sdk_node_publisher.cpp:(.text+0xdd2): undefined reference to `DJI::OSDK::DataBroadcast::getBatteryInfo() const'
dji_sdk_node_publisher.cpp:(.text+0xe64): undefined reference to `DJI::OSDK::DataBroadcast::getBatteryInfo() const'
dji_sdk_node_publisher.cpp:(.text+0xf2f): undefined reference to `DJI::OSDK::DataBroadcast::getStatus() const'
dji_sdk_node_publisher.cpp:(.text+0xfc7): undefined reference to `DJI::OSDK::DataBroadcast::getGimbal() const'
CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_publisher.cpp.o: In function `DJISDKNode::publish50HzData(DJI::OSDK::Vehicle*, DJI::OSDK::RecvContainer, void*)':
dji_sdk_node_publisher.cpp:(.text+0x1c25): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x1c8f): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x1cf9): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x1d63): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
dji_sdk_node_publisher.cpp:(.text+0x1dcd): undefined reference to `DJI::OSDK::DataBroadcast::getRC() const'
CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_publisher.cpp.o:dji_sdk_node_publisher.cpp:(.text+0x1e23): more undefined references to `DJI::OSDK::DataBroadcast::getRC() const' follow
collect2: error: ld returned 1 exit status
make[2]: *** [/home/user/directory/devel/.private/dji_sdk/lib/dji_sdk/dji_sdk_node] Error 1
make[1]: *** [CMakeFiles/dji_sdk_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/user/directory/build/dji_sdk; catkin build --get-env dji_sdk | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

mithunvarma, 2017-08-15:
<solved>
required  these header files in dji_sdk_node_publisher.cpp

#include <djiosdk/dji_control.hpp>
#include <djiosdk/dji_status.hpp>
#include <djiosdk/dji_version.hpp>

or i think you can add 
#include <dji_sdk/dji_sdk.h>

rohitsantdji, 2017-08-15:
We have tested and can successfully build the latest [3.3.1 tag](https://github.com/dji-sdk/Onboard-SDK-ROS/tree/3.3.1) of OSDK-ROS after `sudo make install`ing the latest 3.3 branch of the core OSDK. Note that you need the absolute latest commit on both branches - even having one commit older on either branch will result in a build failure.

mithunvarma, 2017-08-15:
so I tried in a different computer with latest 3.3.1 of both packages, now im getting this error,

(py27) lonai@ROSLonai:~/lonsad/src/Onboard-SDK-3.3/osdk-core$ cmake .
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lonai/lonsad/src/Onboard-SDK-3.3/osdk-core
(py27) lonai@ROSLonai:~/lonsad/src/Onboard-SDK-3.3/osdk-core$ sudo make install .
Scanning dependencies of target djiosdk-core
[  3%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_error.cpp.o
[  7%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_waypoint.cpp.o
[ 10%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_gimbal.cpp.o
[ 14%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_hardware_sync.cpp.o
[ 17%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_vehicle.cpp.o
[ 21%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_mission_manager.cpp.o
[ 25%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_command.cpp.o
[ 28%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_camera.cpp.o
[ 32%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_subscription.cpp.o
[ 35%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_broadcast.cpp.o
[ 39%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_version.cpp.o
[ 42%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_mfio.cpp.o
[ 46%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_hotpoint.cpp.o
[ 50%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_mobile_communication.cpp.o
[ 53%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_control.cpp.o
[ 57%] Building CXX object CMakeFiles/djiosdk-core.dir/api/src/dji_ack.cpp.o
[ 60%] Building CXX object CMakeFiles/djiosdk-core.dir/protocol/src/dji_aes.cpp.o
[ 64%] Building CXX object CMakeFiles/djiosdk-core.dir/protocol/src/dji_open_protocol.cpp.o
[ 67%] Building CXX object CMakeFiles/djiosdk-core.dir/hal/src/dji_thread_manager.cpp.o
[ 71%] Building CXX object CMakeFiles/djiosdk-core.dir/hal/src/dji_log.cpp.o
[ 75%] Building CXX object CMakeFiles/djiosdk-core.dir/hal/src/dji_memory.cpp.o
[ 78%] Building CXX object CMakeFiles/djiosdk-core.dir/hal/src/dji_hard_driver.cpp.o
[ 82%] Building CXX object CMakeFiles/djiosdk-core.dir/utility/src/dji_circular_buffer.cpp.o
[ 85%] Building CXX object CMakeFiles/djiosdk-core.dir/utility/src/dji_singleton.cpp.o
[ 89%] Building CXX object CMakeFiles/djiosdk-core.dir/platform/default/src/dji_memory_default.cpp.o
[ 92%] Building CXX object CMakeFiles/djiosdk-core.dir/platform/linux/src/linux_serial_device.cpp.o
[ 96%] Building CXX object CMakeFiles/djiosdk-core.dir/platform/linux/src/posix_thread_manager.cpp.o
[100%] Building CXX object CMakeFiles/djiosdk-core.dir/platform/linux/src/posix_thread.cpp.o
Linking CXX static library libdjiosdk-core.a
[100%] Built target djiosdk-core
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/cmake/djiosdk/DJIOSDKConfig.cmake
-- Installing: /usr/local/lib/cmake/djiosdk/DJIOSDKConfigVersion.cmake
-- Installing: /usr/local/lib/cmake/djiosdk/djiosdkTargets.cmake
-- Installing: /usr/local/lib/cmake/djiosdk/djiosdkTargets-release.cmake
-- Installing: /usr/local/lib/libdjiosdk-core.a
-- Up-to-date: /usr/local/include/djiosdk/dji_vehicle.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_status.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_mobile_communication.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_waypoint.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_hardware_sync.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_command.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_broadcast.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_hotpoint.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_telemetry.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_control.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_mfio.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_camera.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_version.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_subscription.hpp
-- Up-to-date: /usr/local/include/djiosdk/dji_gimbal.hpp
[... 48 lines cut ...]
Starting  >>> dji_sdk                                                                                                                                    
_________________________________________________________________________________________________________________________________________________________
Errors     << dji_sdk:cmake /home/lonai/lonsad/logs/dji_sdk/build.cmake.003.log                                                                          
CMake Error at /opt/ros/indigo/share/catkin/cmake/assert.cmake:20 (message):

  Assertion failed: file
  '/home/lonai/lonsad/src/Onboard-SDK-ROS-3.3/dji_sdk/msg/WaypointList.msg'
  does not exist.  Message: message file not found

Call Stack (most recent call first):
  /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:113 (assert_file_exists)
  CMakeLists.txt:51 (add_message_files)

[... cut ...]
[build]   Warnings:  None.                                                                                                                               
[build]   Abandoned: 1 packages were abandoned.                                                                                                          
[build]   Failed:    1 packages failed.                                                                                                                  
[build] Runtime: 0.9 seconds total.                                                                                                                      
(py27) lonai@ROSLonai:~/lonsad$

amenonDJI, 2017-08-22:
@mithunvarma Closing this issue due to inactivity. Please reopen if you continue to face problems.

closed 2017-08-22 by amenonDJI