dji-sdk/Onboard-SDK-ROS #120: Horizontal command co-ordinate frames may need clarification or there is a bug?

chutsu, 2017-08-16:
**Onboard-SDK-ROS version: 3.2**

I'm trying to give specific attitude commands (roll, pitch, yaw and thrust) to the DJI N3, I'm achieving this by setting a control mode byte of `0x20`

```
  //  DJI Control Mode Byte
  //
  //    bit 7:6  0b00: HORI_ATTI_TILT_ANG
  //             0b01: HORI_VEL
  //             0b10: HORI_POS
  //
  //    bit 5:4  0b00: VERT_VEL
  //             0b01: VERT_POS
  //             0b10: VERT_THRUST
  //
  //    bit 3    0b0: YAW_ANG
  //             0b1: YAW_RATE
  //
  //    bit 2:1  0b00: horizontal frame is ground frame
  //             0b01: horizontal frame is body frame
  //
  //    bit 0    0b0: non-stable mode
  //             0b1: stable mode
  //
  //  We used:
  //
  //    HORIZ_ATTI_TILT_ANG
  //    VERT_THRUST
  //    YAW_ANG
  //    ground frame
  //    non-stable mode
  //
  //  ends up being: 0b00100000 -> 0x20
```
If I set the horizontal frame to be "ground frame" I noticed that if I set a roll command at 1 degree, but yaw 90 degrees the quadrotor actually pitches rather rolls, this tells me that there is a hidden transform that rotates roll and pitch given a yaw. I am not sure this is desired behaviour because absolute roll, pitch and yaw angles should be relative to the ground frame.

I found a _**fix**_ to this problem by setting the horizontal frame to "body frame" which changes the control byte from `0x20` to `0x22`, but then a body frame doesn't work as desired either for if I give it a roll of 1 degree in the body frame, I would expect the quadrotor to keep spinning in the roll direction, since the roll command is relative to the body frame.

In conclusion there needs to be more clarification in the ROS wiki about what the "horizontal frame" command means. **I have not had the chance to test this behaviour in the latest SDK 3.3.1**  so I cannot confirm it exists in the latest SDK.

Here is a script `test_script.py` (see below) I wrote to test the attitude commands, assumes you're using **SDK version 3.2**:

```
#!/usr/bin/env python
from time import sleep

from dji_sdk.dji_drone import DJIDrone

if __name__ == "__main__":
    drone = DJIDrone()

    sleep(1)
    print("Requesting DJI Control!")
    drone.request_sdk_permission_control()
    sleep(1)
    print("Got control!")

    print("Attitude control!")
    while True:
        roll = 1.0
        pitch = 0.0
        thrust = 35.0
        yaw = -90.0
        # control_byte = 0x22  # body frame attitude command (works as intended)
        control_byte = 0x20  # ground frame attitude command (fails)
        drone.attitude_control(control_byte, roll, pitch, thrust, yaw)
```

**Test procedure:**

1. Power on the DJI N3
2. Connect DJI assistant to DJI N3 and launch the simulator
3. `roslaunch dji_sdk sdk_manifold.launch` (assuming you have proper app id and key)
4. Arm the quadrotor in the DJI assistant simulator
5. launch the test script above by typing `python test_script.py`

chutsu, 2017-08-16:
Having thought about this problem further, if I were to debug code I would look at the code logic of the horizontal commands more closely, because if the horizontal component is defined to be ground frame, to rotate both the first two components of the horizontal component by the yaw makes perfect sense if they are ground frame velocity and position (or `HORI_VEL` and `HORI_POS` in the control byte), but it doesn't make sense in attitude mode `HORI_ATTI_TILT_ANG`, so the code **might be missing a logic to skip the yaw rotation** to bring the horizontal command from the ground frame to the body frame.

andre-nguyen, 2017-08-16:
Maybe these issues can help you out:

* https://github.com/dji-sdk/Onboard-SDK-ROS/issues/25 Control mode clarification
* https://github.com/dji-sdk/Onboard-SDK-ROS/issues/32 Horizontal frame clarification

amenonDJI, 2017-08-16:
Hi @chutsu 

Tested this on OSDK 3.2 modifying the demo and the aircraft behaves as expected, that is, the aircraft rolls in body frame by 10 degrees and points -90 degrees in yaw as it ascends in altitude. 

`
                for(int i = 0;i < 150;i++)
                {
                    drone->attitude_control( Flight::HorizontalLogic::HORIZONTAL_ANGLE |
                            Flight::VerticalLogic::VERTICAL_THRUST |
                            Flight::YawLogic::YAW_ANGLE |
                            Flight::HorizontalCoordinate::HORIZONTAL_GROUND |
                            Flight::SmoothMode::SMOOTH_DISABLE,
                            10, 0, 35, -90 );  // Tested with 1 degree roll and 10 degree roll
                    usleep(20000);
                }

                break;
`
Take a look at the links from @andre-nguyen. It will be worthwhile to modify the demo and test if you see the same behavior.

closed 2017-12-05 by chutsu