dji-sdk/Onboard-SDK-ROS #122: SDK Access disabled and settings reset randomly on M600

fong7x7, 2017-08-22:
When testing the OSDK on the M600, the M600 would sometimes boot up with SDK control disabled after the setting had been enabled on DJI Assistant 2.  In addition to the SDK control being disabled, the M600 would also disable multiple flight modes.  Both the SDK control and the multiple flight modes can be manually re-enabled, but it's frustrating to have the settings randomly reset. Is there any way this can be fixed?

amenonDJI, 2017-08-22:
@fong7x7 When you upload new firmware to the Aircraft, the SDK Control is disabled. Is this behavior seen at other times as well? If so, could you provide the version of your M600 firmware and Assistant2.

fong7x7, 2017-08-23:
@amenonDJI We haven't updated the firmware for the M600 in a while.  The M600 firmware version is 1.0.1.20 and the DJI Assistant 2 is 1.0.9-2.

amenonDJI, 2017-08-23:
@fong7x7 We have tested M600 firmware version 1.0.1.20 and have never run into this issue. Could you upgrade to the latest Assistant and confirm if you still see this behavior?

fong7x7, 2017-08-29:
@amenonDJI We just set the toggle with the latest version of Assistant and we are still experiencing the issue. We have an onboard computer that connects to the M600 through the API port that loses data communication when the sdk disables.  After we notice that the onboard computer is not connecting to the M600, that's when we check Assistant to see that the sdk was disabled.  Based on this, we believe that the issue is happening on the M600 and not with Assistant.

closed 2021-06-25 by dji-kyle