dji-sdk/Onboard-SDK-ROS #123: RC topic not publishing correct data

fong7x7, 2017-08-31:
On Onboard-SDK-ROS version 3.3 using an A3 on version v1.7.1.5 with controller v1.2.10, the topic "dji_sdk/rc" does not publish the correct frequencies for the controller values.  The values are set at a default frequency and do not change when the controller inputs change.

fong7x7, 2017-08-31:
I've found the source of the issue.  
It appears that the rc topic has been changed to use vehicle->broadcast instead of vehicle->subscribe.  This can be found in dji_sdk_node_publisher.cpp line 333 to line 390.  

From my testing, the vehicle->broadcast call does not update real time with the changes on the controller causing the topic to not publish the correct data.  However, the original subscribe was removed because of this comment:
"When using SBUS remote controllers, the rc.mode does not properly reflect the rc's input."

Can the code be updated back to use vehicle->subscribe with support for SBUS?

zliDJI, 2017-08-31:
Hello, even with broadcast, you should be able to receive RC data if you set the frequency correctly to 50 Hz in DJI assistant 2.

fong7x7, 2017-09-01:
@zliDJI Is it the Remote Control Channel Data setting? I currently have that setting set to 50 Hz and it still wasn't broadcasting the correct controller inputs.  Note: it does broadcast and publish data at 50hz, but it only broadcasts the initial controller state does not update to any real time controller inputs.

asiron, 2018-06-15:
https://github.com/dji-sdk/Onboard-SDK-ROS/blob/f3b136eb769e23f9128ce77babc9d7584659b707/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L414-L419

Are you planning on fixing it in the near future ?

We're using the **Subscription** mode from the SDK, since it provides additional topics such as `/dji_sdk/acceleration_ground_fused`, `/dji_sdk/angular_velocity_fused`, besides the fact that the old **Broadcast** mode is getting deprecated.

rohitsantdji, 2018-06-15:
Hi @asiron we're happy to say that OSDK 3.7 will introduce a wide range of RC-related improvements, including this one. At release, we will support A3, N3 and M210 for these improvements, but the M600/Pro will get support with a firmware update some time after the release of OSDK 3.7. The release timeframe for OSDK 3.7 is tentatively being planned towards the end of July.

closed 2021-06-25 by dji-kyle