dji-sdk/Onboard-SDK-ROS #124: Drone allows takeoff when control is not available

rauhit, 2017-09-01:
I am running a hardware in the loop simulation with an M600.

When I run the demo flight control node with the PAF switch in the A position the SDK fails to get control, but the aircraft still allows the takeoff command.
```
$ rosrun dji_sdk_demo demo_flight_control 
[ERROR] [1504271074.914197952]: obtain control failed!
[ INFO] [1504271074.915005103]: A3/N3 taking off!
[ INFO] [1504271075.152854919]: Motor Spinning ...
[ INFO] [1504271078.053235448]: Ascending...
[ INFO] [1504271079.875531091]: Successful takeoff!
```

The RC message is: 
`axes: [0.0, 0.0, 0.0, 0.0, 0.0, -5000.0]`

Is there a any flight mode where the SDK has no control over the aircraft? 

Thanks,

rauhit, 2017-09-01:
Firmware version on DJI M600: 1.0.1.60
SDK version: 3.3

frontw, 2017-09-10:
Confirm this

rohitsantdji, 2017-09-11:
Please see the reply on #139.

On the M600, there is no flight mode where the SDK cannot obtain control; however after each flight mode switch toggle, the SDK must request control. This is a behavior your code can shape as per your requirements.

Also from the Control Authority Guide (mentioned on the linked issue) - takeoff and other similar flight commands (go home, landing) are not gated by control authority, and may be called without obtaining control authority.

closed 2021-06-25 by dji-kyle