dji-sdk/Onboard-SDK-ROS #126: Wrong topic fpr velocity control

tompe17, 2017-09-01:
In demo_flight_control.cpp the line
`ctrlVelYawRatePub = nh.advertise<sensor_msgs::Joy>("dji_sdk/flight_control_setpoint_generic", 10);`
should be
`ctrlVelYawRatePub = n.advertise<sensor_msgs::Joy>("dji_sdk/flight_control_ENUvelocity_yawrate", 10);`

zliDJI, 2017-09-05:
Hi, @tompe17  Thanks for the finding. The topic type of ctrlVelYawRAtePub was set to flight_control_setpoint_generic intentionally. But when we sent the command, there was a bug. Line 172-176, and line 182, should have been in "else if(break_counter > 0)", i.e.,

```
  else if(break_counter > 0)
  {
  sensor_msgs::Joy controlVelYawRate;

    uint8_t flag = (DJISDK::VERTICAL_VELOCITY   |
                    DJISDK::HORIZONTAL_VELOCITY |
                    DJISDK::YAW_RATE            |
                    DJISDK::HORIZONTAL_GROUND   |
                    DJISDK::STABLE_ENABLE);

    controlVelYawRate.axes.push_back(0);
    controlVelYawRate.axes.push_back(0);
    controlVelYawRate.axes.push_back(0);
    controlVelYawRate.axes.push_back(0);
    controlVelYawRate.axes.push_back(flag);

    ctrlVelYawRatePub.publish(controlVelYawRate);
    break_counter++;
    return;
  }
  else //break_counter = 0, not in break stage
  {
    sensor_msgs::Joy controlPosYaw;
    controlPosYaw.axes.push_back(xCmd);
    controlPosYaw.axes.push_back(yCmd);
    controlPosYaw.axes.push_back(zCmd);
    controlPosYaw.axes.push_back(yawDesiredRad);
    ctrlPosYawPub.publish(controlPosYaw);
  }
```
We'll push a hotfix later. Thanks!

closed 2017-09-09 by zliDJI

tompe17, 2017-09-09:
For me the name ctrlVelYawRatePub was the confusing thing. I assumed from the name how it should work when I copied some code from the sample.

commit 5e6f683f (2017-09-09): Local position topic and bug fixes:
closes #121, closes #125, closes #126, closes #133