dji-sdk/Onboard-SDK-ROS #127: Take-off position on a topic or through a service call?

tompe17, 2017-09-01:
As I understand it the altitude you specify for control is relative to the take-off position. But how do I get that position? I have looked for service call to get it or a topic but cannot find anything.

Would it be possible to have a topic with that position or a service call to get it?

I need it to be able to specify an absolut position to fly to where the altitude is specified relative to some geoid.

zliDJI, 2017-09-07:
Hi, @tompe17 . Can you record the current position (lat, lon, alti) before you take off?

tompe17, 2017-09-09:
Yes, it is possible for me to record the position and use that but such a solution will not be as robust as if the data come from the actual controller. I have already started with implementing such a solution.

But the proper solution would be to make it possible to relate height above takeoff position to a global position without having to record specific information when doing takeoff but just using the available topics from the dji_sdk_node. 

It seems a bit strange to me that if you have two vehicles and get the topics from them it is not possible to see if they are on the same altitude (relative to error in sensors).

amenonDJI, 2018-01-24:
@tompe17 Please reference [this](https://github.com/dji-sdk/Onboard-SDK/wiki/Altitude-Reference-for-different-topics) wiki page for more information regarding your concern.

closed 2021-06-25 by dji-kyle