dji-sdk/Onboard-SDK-ROS #128: Forced update broke software

ben-horrocks, 2017-09-01:
We built software to run on SDK version 3.2. After we got it running, our platform was forced to update to SDK version 3.3 without our consent, which broke our software three days before a big demo. Forcing updates without backwards compatibility to existing contracts is a poor development practice or a bug.

zliDJI, 2017-09-01:
Hi, how was your system forced to upgrade? You should be able to use DJI Assistant 2 to downgrade your firmware to 1.0.1.20, which supports SDK 3.2.

rohitsantdji, 2017-09-02:
Hi @ben-horrocks - we apologize for this experience. Are you using M600 or M600 pro? We are investigating a recent change in the M600 firmware upgrade logic regarding the extent of what this affects. We'll report back on this issue early next week with some potential solutions.

sdo91, 2017-09-05:
Hi, coworker of @ben-horrocks here. We are using the M600 pro. We were using the latest firmware compatible with OSDK 3.2, but when the new M600 firmware was released 6 days ago, the dji app would not let us fly without updating to the new firmware (which only supports OSDK 3.3).

zliDJI, 2017-09-05:
@sdo91  If you haven't upgraded to the latest FW, and is still on 1.0.1.20, you should be able to take of your drone by tapping the "Drone is locked" message in DJI go. If you've already upgraded to 1.0.1.60, we are working on a solution.

rohitsantdji, 2017-09-05:
Hi @sdo91 @ben-horrocks. To add to @zliDJI's comment - we've set up a procedure for recovering the old firmware on a temporary basis, until you are able to migrate your code to OSDK 3.3. To access this procedure, please **email dev@dji.com requesting an OSDK-related firmware downgrade.** 

The new 3.3 firmware brings safety updates regarding batteries and regulatory updates regarding no-fly-zones, so we recommend upgrading your codebase to 3.3 APIs as soon as possible.

jswoboda, 2017-10-04:
Hi I'm also having similar issues using the python api with 3.2. Will that be updated?

closed 2021-06-25 by dji-kyle