dji-sdk/Onboard-SDK-ROS #134: dji_sdk_demo and other ROS packages, that use ROS interface to DJI SDK should not depend on DJIOSDK

frontw, 2017-09-05:
We should have ROS interface to all DJI OSDK features trough topics and services.
Why dji_sdk_demo depend on DJIOSDK as linked library?
In CMakeLists.txt following line should be removed
find_package(DJIOSDK REQUIRED)
and ${DJIOSDK_LIBRARIES} in target_link_libraries.
But if we do so, we'll get error: undefined reference to `DJI::OSDK::Version::FW(unsigned char, unsigned char, unsigned char, unsigned char)'.
This is because we include #include "dji_sdk/dji_sdk.h" to have some flags.
But dji_sdk/dji_sdk.h includes djiosdk/dji_version.hpp in turn, which has not only flags and structs definitions, but Version class with FW function declaration.

zliDJI, 2017-09-07:
You are right, user/client code should talke to dji_sdk_node through only services and topics. We'll look into how to fix this. THanks!

Tcloser, 2020-02-24:
> This is because we include #include "dji_sdk/dji_sdk.h" to have some flags.
Thank u !

closed 2021-06-25 by dji-kyle