dji-sdk/Onboard-SDK-ROS #135: How could I found drone is controled by SDK API?

frontw, 2017-09-06:
My app logic is to check, if drone is in SDK control mode and if not, request sdk permission control.
Display mode has MODE_NAVI_SDK_CTRL, but this mode is not set after successful /dji_sdk/sdk_control_authority call, but only when I publish first msgs at /dji_sdk/flight_control_setpoint_ENUposition_yaw topic after it.

zliDJI, 2017-09-07:
Can you just request sdk control regardless of the current status? If you are already in control, the request will do nothing.

frontw, 2017-09-10:
Thanks, SDK Info updates promptly.

Btw, is it OK, that I could takeoff from 
`rosservice call /dji_sdk/drone_task_control "task: 4" `
without getting SDK control
`rosservice call /dji_sdk/sdk_control_authority "control_enable: 1"`
?

frontw, 2017-09-10:
Oh. Update. sdk_info publishes device_status: 2 and flight_status: 1, but I still could control with remote controller.
It doesn't work.

rohitsantdji, 2017-09-11:
This is expected behavior! Please see the [Control Authority Guide](http://developer.dji.com/onboard-sdk/documentation/guides/component-guide-control-authority.html). 

I think the key point to understand is that **having** control authority is not exclusive. If the SDK has control authority, then *as long as the SDK is not actually sending control commands* the RC can still control it. Think of the control authority as a necessary step along the way to controlling the drone. 

As per the flowchart in the doc, you will be unable to use the RC only if the state transitions into "Sending movement control commands, or starting a mission".

closed 2021-06-25 by dji-kyle