dji-sdk/Onboard-SDK-ROS #136: Wrong unit (rad * 10 vs deg * 10) in gimbal_angle_cmd and gimbal_speed_cmd

frontw, 2017-09-07:
http://wiki.ros.org/dji_sdk says:

> /dji_sdk/gimbal_angle_cmd (dji_sdk/Gimbal)
> 
>     Gimbal control command: Controls the Gimbal roll pitch and yaw angles (unit: **0.1 deg**). mode: 0 - incremental control, the angle reference is the current Gimbal location. 1 - absolute control, the angle reference is related to configuration in DJI Go App. 
> 
> /dji_sdk/gimbal_speed_cmd (geometry_msgs/Vector3Stamped)
> 
>     Gimbal speed command: Controls the Gimbal rate of change for roll pitch and yaw angles (unit: 0**.1 deg/sec**). 

But in dji_sdk_node_subscriber.cpp we could see:

```
  //! OSDK takes 0.1 deg as unit
  angle_data.roll     = RAD2DEG(msg->roll)*10;
  angle_data.pitch    = RAD2DEG(msg->pitch)*10;
  angle_data.yaw      = RAD2DEG(msg->yaw)*10;
  vehicle->gimbal->setAngle(&angle_data);
```

and
```
  //! OSDK takes 0.1 deg as unit
  speed_data.roll  = RAD2DEG(msg->vector.x)*10;
  speed_data.pitch = RAD2DEG(msg->vector.y)*10;
  speed_data.yaw   = RAD2DEG(msg->vector.z)*10;
  vehicle->gimbal->setSpeed(&speed_data);
```

So dji_sdk assumes data in gimbal_angle_cmd and gimbal_speed_cmd is in **rad** and **rad/s**.

rohitsantdji, 2017-09-07:
The code is made to be compatible with REP-103, so the unit of specifying angles (radians) and angular rate (radians/sec) are correct in the code. The documentation will be updated to reflect this.

closed 2017-09-07 by rohitsantdji

frontw, 2017-09-09:
You reference REP-103, but it is not strong requirement for all cases. Please, take into account the fact, that ROS gives you ability to watch all data published in topic in human-frendly style usually. I mean, you could always rostopic echo /gps_position for example and see, if data is correct. Publishing radians instead degrees for such data as gimbal angles, latitude and longitude is not user-friendly.
Hope, you'll agree.
Thanks!