dji-sdk/Onboard-SDK-ROS #137: Gimbal control and feedback data seems to have different reference

frontw, 2017-09-07:
Yaw data seems to have some bias.

When I set yaw to 0, I get -59 deg, when I set 20, I get -39. See screenshots. Roll and pitch are controlled normally. 

[image]
[image]

I think, there is some problem with gimbal reference, it's different for control and feedback topic, but should be the same.

Note: I removed DEG2RAD according to #136.

rohitsantdji, 2017-09-07:
The yaw angle reported by the gimbal is in an absolute (Earth-fixed) frame - so the values in yaw are with respect to absolute north.

frontw, 2017-09-07:
Ok, then control should be in same frame, yes?

zliDJI, 2017-09-07:
For control, there are multiple modes, with angle meaning different depending on the modes, please refer to the docs:

https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1Gimbal_1_1AngleData.html

frontw, 2017-09-07:
Please, review my screenshots. If I give an angle of 20 degrees, for example, should I get the same (approximately) angle in feedback (as it is for roll and pitch, but not yaw)? Or I need to give control with one reference and get result with another?

zliDJI, 2017-09-07:
You set bit 0 of mode to 1, which according to documentation in my link,  "Absolute control, the angle reference is related to configuration in DJI Go App". 

However, in DJI Go App, the default configuration is "Follow", meaning that the Yaw is with respect to body frame (see the docs in my link), so when you set yaw to 0 degree, gimbal will point to the nose of the drone.

zliDJI, 2017-09-08:
Hi @frontw, we verified your finding and found that in "free" mode, the gimbal is not behaving the way as documented. The gimbal is moving with relative angles in both "free" and "follow" mode.
 We'll investigate the issue internally and get back to you. Thanks and sorry about the issue.

closed 2021-06-25 by dji-kyle