dji-sdk/Onboard-SDK-ROS #138: Please, consider again your idea to publish all angle data in radians instead degrees

frontw, 2017-09-09:
I saw in last commit you change even latitude to radians.
You reference REP-103, but it is not strong requirement for all cases. Please, take into account the fact, that ROS gives you ability to watch all data published in topic in human-frendly style usually. I mean, you could always rostopic echo /gps_position for example and see, if data is correct. Publishing radians instead degrees for such data as gimbal angles, latitude and longitude is not user-friendly. 

Also, you publish NavSatFix incorrect now, look at http://docs.ros.org/jade/api/sensor_msgs/html/msg/NavSatFix.html

```
# Latitude [degrees]. Positive is north of equator; negative is south.
float64 latitude

# Longitude [degrees]. Positive is east of prime meridian; negative is west.
float64 longitude
```

Hope, you'll agree. 
Thanks!

zliDJI, 2017-09-12:
Do you want to publish angles on degree or radians?  

Currently we **are** publishing lat, long, gimbal angles in degree. Could you point:
 - Where in our last commit we change lat to radians? 
 - Which part of our NavSatFix is wrong?

Thanks!

frontw, 2017-09-12:
Here #136 was noted, that gimbal data is published in radians instead degrees

frontw, 2017-09-12:
Couldn't confirm about NavSatFix, sorry.

Leaving gimbal angles question. According to https://github.com/dji-sdk/Onboard-SDK-ROS/issues/136#issuecomment-327857294, it is in radians, I propose to use degrees for it.

closed 2021-06-25 by dji-kyle