dji-sdk/Onboard-SDK-ROS #139: M600Pro SDK control and Flight Mode Switch

frontw, 2017-09-10:
M600Pro Manual says:
> The external device will be able to control the aircraft only if the Flight Mode Switch on the remote controller is toggled to F-mode. 
But I see, that SDK program could get control in P and F mode both, do I could fly only in ATT mode remotely when testing? It's dangerous, I guess.
What I'm doing wrong?

Also after switch to A-mode and returning to F or P, SDK control couldn't be acquired again for some reason.

snakehaihai, 2017-09-11:
haha same here. In opened issue turn left is my error report. haha. My diagnose is their A3 MCU has some bug. And they didn`t open it like APM firmware. so no way for us to debug it.  But 3.3 is supposed to work like the APM rosserial subscribe to /cmd_vel sort of thing, it is already an improvement.

rohitsantdji, 2017-09-11:
The behavior of SDK control has changed since the M600 Pro manual was released, and is kept up-tp date in the [Control Authority Guide](http://developer.dji.com/onboard-sdk/documentation/guides/component-guide-control-authority.html). 

From the guide: 
>When a remote controller is used with M600 or M600 Pro, all flight mode switch positions are P mode always. Similar to Multiple Flight Modes disabled for A3/N3, the switch position only needs to be **changed** to regain control with the remote controller. **Control is only temporarily regained**, and logic in the onboard computer should monitor the remote controller flight mode switch and stop sending commands to the aircraft if the user needs to use the switch to permanently regain control.

Can you confirm that the behavior you are seeing matches with this expectation? 

1. If you toggle flight mode from F or P to any mode, the SDK should lose control
2. In any mode after this, the SDK should be able to obtain control.

snakehaihai, 2017-09-11:
see [https://github.com/dji-sdk/Onboard-SDK-ROS/issues/140](url)

frontw, 2017-09-11:
Thanks for this link!

closed 2017-09-11 by frontw