dji-sdk/Onboard-SDK-ROS #140: Manual override did not response

snakehaihai, 2017-09-11:
Hi
I`m using new 3.3 api to control UAV autonomously using vision. 
for 3.2 control and bypass is fine. I can control UAV using Lidar and sent out control signal at P mode. anything wrong just switch to F mode. The only problem is with IMU update frequency is too low that I can not run vision-inertial based navigation with it. 

In 3.3 IMU is pub at 400 hz which is ideal. When shit happens at P mode, I switch to F mode. However, the drone does not regain control from RC. May I ask how to solve this issue?
I`m using dji sdk 3.3 and lightbridge 2. In dji_sdk terminal only shows request control, never saw release control.

Regards
Shenghai Yuan

rohitsantdji, 2017-09-11:
Please see #139 .

snakehaihai, 2017-09-11:
confirm no use.
when shit happens. UAV carrying a payload of 1kg fly with 80-degree pitch angle flying towards the wall. 
I`m the main developer, In software control mode, for a brief time, I have control.  I tried to release software control-> no use.  Then i call service to release contorl. So that i can test emergency override

my assistant server as the Secondary pilot tried to disarm drone in both P mode and F mode. no use
He mention to me for a short period of time, He has control. after that then lost, and drone fly 80 degrees to the wall. 
We were panic. lucky drone is on our test rig and we know for 1st a few time shit happens.
Then we didn`t dare to do any more test. but to test out software response

and we notice that api can have control no matter in P or F mode for some reason. Will confirm this and take a picture to show you tmr

rohitsantdji, 2017-09-11:
Are you using an A3/N3 on a custom vehicle, or an M600? Please note that as the docs on control authority say, the behavior is different for both cases. 

> But in flight, after I`m in API mode and switch back. and we notice that API can have control no matter in P or F mode for some reason. 

As i mentioned before, this is expected as per the docs. You should monitor the mode switch positions and send release/obtain control commands accordingly - here is a small snippet that works:

```
    //! Check RC commands for release control
    curRC = vehicle->broadcast->getRC();
    if (curRC.mode == 10000 && (!isRCToggle))
    {
      isRCToggle = true;
      DSTATUS("Releasing control..");
      vehicle->releaseCtrlAuthority(1);
    }
    if (curRC.mode == -10000 && (isRCToggle))
    {
      isRCToggle = false;
      DSTATUS("Releasing control..");
      vehicle->releaseCtrlAuthority(1);
    }
```
isRCToggle is a bool that maintains state.

For the other issue regarding dji_sdk/flight_control_setpoint_generic, please open a new issue since it is unrelated to this discussion and will help others if the two are kept separate.

snakehaihai, 2017-09-12:
With lightbridge controller off->   called service vehicle->obtainCtrlAuthority  ack=2
                                                    called service vehicle->releaseCtrlAuthority ack=1
mode switch = P->   called service vehicle->obtainCtrlAuthority  ack=2
                                                    called service vehicle->releaseCtrlAuthority ack=1
mode switch = A->   called service vehicle->obtainCtrlAuthority  ack=2
                                                    called service vehicle->releaseCtrlAuthority ack=1
mode switch = F->   called service vehicle->obtainCtrlAuthority  ack=2
                                                    called service vehicle->releaseCtrlAuthority ack=1

Funny thing is, I remember in some cases, the red warning pop up say I can't obtain control in the dji driver. Just could not remember how I did it

snakehaihai, 2017-09-12:
ok previous flight was not turn on enable multple flight mode flag.
but trickly thing is once i turn the flag on. restart drones follow thing happens

called vehicle->obtainCtrlAuthority
2
called vehicle->obtainCtrlAuthority
2
called vehicle->obtainCtrlAuthority
2
called vehicle->releaseCtrlAuthority
1
called vehicle->releaseCtrlAuthority
1
called vehicle->releaseCtrlAuthority
1
called vehicle->releaseCtrlAuthority
1
called vehicle->obtainCtrlAuthority
2
called vehicle->obtainCtrlAuthority
2
called vehicle->obtainCtrlAuthority
2
called vehicle->obtainCtrlAuthority
2
called vehicle->obtainCtrlAuthority
2
called vehicle->releaseCtrlAuthority
1
called vehicle->releaseCtrlAuthority
1
called vehicle->releaseCtrlAuthority
1
called vehicle->releaseCtrlAuthority
0

STATUS/1 @ getErrorCodeMessage, L590: 
STATUS/1 @ getCMDIDSetControlMSG, L724: RC_NEED_MODE_P

STATUS/1 @ getCMDIDSetControlMSG, L731: RC_MODE_ERROR
called vehicle->releaseCtrlAuthority
0

STATUS/1 @ getErrorCodeMessage, L590: sdkCtrlAuthorityCallback
STATUS/1 @ getCMDIDSetControlMSG, L724: RC_NEED_MODE_P

STATUS/1 @ getCMDIDSetControlMSG, L731: RC_MODE_ERROR
called vehicle->releaseCtrlAuthority
0

STATUS/1 @ getErrorCodeMessage, L590: sdkCtrlAuthorityCallback
STATUS/1 @ getCMDIDSetControlMSG, L724: RC_NEED_MODE_P

STATUS/1 @ getCMDIDSetControlMSG, L731: RC_MODE_ERROR
called vehicle->releaseCtrlAuthority
0

STATUS/1 @ getErrorCodeMessage, L590: sdkCtrlAuthorityCallback
STATUS/1 @ getCMDIDSetControlMSG, L724: RC_NEED_MODE_P

STATUS/1 @ getCMDIDSetControlMSG, L731: RC_MODE_ERROR
called vehicle->obtainCtrlAuthority
0

```

which is strange. for 1st round of P A F switching. software shows no error in obtain control and release control. for subsequent round. all error pops up.

snakehaihai, 2017-09-12:
I verified again. upon start, UAV API can control, no matter what state it is. only upon start

snakehaihai, 2017-09-14:
Hi i was flying and it went out of control. We verified the problem like below.
Although F mode does not allow API control. but even in F mode, RC is not working as well. 

the problem can be reproduced as follows
1. arm the drone in P mode. software can obtain and release the control.
2. while software obtain control,  switch from P to F mode.
it appear to be both software and lightbriedge RC dont have any control at all. emergency by pass is not working

only switch from F to P again. then use software control. then i can land the drone.

snakehaihai, 2017-09-18:
dji error insufficient payload..then uav go to full throttle and pitch back mode, F mode by pass not working, API by pass not working. how to solve this

rohitsantdji, 2017-09-18:
Is your multiple flight modes switch enabled in Assistant? If yes, what are your switch configurations?

snakehaihai, 2017-09-21:
Yes I have to enable switch in order to achieve bypass right?  Mode are P A S. and in simulation everything is ok.

snakehaihai, 2017-09-22:
To summary the errors.  
1.Error insufficient hover thrust blah blah blah -> A3 need enough payload for drone not to fucked up. 2.Possible multiple control mode problems -> Now i`m using P A G mode. 
3.P mode disables paradox -> From the manual when P is disabled, rc temporarily regained. I was sending command even it is 0 0 0 0 in regular 10hz basis that cause manual bypass not working. I have to put the publish(vel) inside the callback. as for possible traffic jam inside the callback only can be solved by a sending time filter to limit the number of output at each time.  So for autonomous control bypass in this 3.3 api, try not to publish control at the regular basis. as it might fucked up the bypass

After solved those problems, I managed to get it fly with pure SDK mode

closed 2017-09-22 by snakehaihai