dji-sdk/Onboard-SDK-ROS #142: 3.3 dji_sdk/flight_control_setpoint_generic control referece to 3.1/3.2  /drone->attitude_control(0x0a,  y,x,z,yaw);

snakehaihai, 2017-09-12:
Can I check with you, this dji_sdk/flight_control_setpoint_generic works like typical RC just that control by serial, right? Is it in stable mode? I tried to control this using Logitech joystick. but hard seems only have altutude control. In 3.1/3.2 we were using attitude_control function call with mode 0x0a. Can hold position indoor very well with lidar feedback. 
whats the difference in the new topic based control node in terms of flight control mode?how is it referred to the old way of control?
I feel the default lightbridge RC flight is quite stable. how can I sent command to drone using API to control as stable as lightbridge RC flight? If can u using either touch screen or logitech joystick to control it as stable as  light bridge RC contorl. then i can interface easily with vision based or lidar based guidance program.

zliDJI, 2017-09-12:
As the name indicates, the topic dji_sdk/flight_control_setpoint_generic will cover whatever is capable by the attitude_control function, and cover more modes (such as commanding roll/pitch rates), based on the flag you provide.

I'll suggest that you read our ros wiki for details: http://wiki.ros.org/dji_sdk#Subscribed_Topics

And more details about possible flags: http://wiki.ros.org/dji_sdk#Details_on_flight_control_setpoint

For your question about interface with your guidance program, you can use sdk to read the RC stick, and send control through API.

snakehaihai, 2017-09-12:
I see from the API

Horizontal | description | reference | limit
-- | -- | -- | --
0x00 | Command roll and pitch angle | Ground/Body | 0.611 rad (35 degree)
0x40 | Command horizontal velocities | Ground/Body | 30 m/s
0x80 | Command position offsets | Ground/Body | N/A
0xC0 | Command angular rates | Ground/Body | 5⁄6π rad/s

So I didn`t set it joy[4] which means default 0x00. means altitude hold first

Say for autonomous mode, there is more than 3 command which is executed at the same time. what should i do?
For example,
if I issue throttle adjust to main UAV at my set point from the help of my vicon/camera/lidar.
Then if issue a yaw pitch command to adjust x y position. 
will the later one override the previous command? since position feedback->action are usually issue at the same time. and previous pub only executed for like 50us?  how to execute 4 commands at all time? like the previous control drone->attitude_control?

zliDJI, 2017-09-13:
We don't use the px4 way to generate the command, which takes the horizontal, yaw, and vertical channel separately. Here you have to send the roll pitch yaw, thrust commands in a single message.

snakehaihai, 2017-09-13:
I understand it not px4 way. 
I`m sending command to /dji_sdk/flight_control_setpoint_generic (sensor_msgs/Joy).
However i only feel throttle control. The pitch and roll control are missing.
The format of the joy topic being published is  joy[0]=roll joy[1]=pitch joy[2]=throttle and joy[3]=yaw.
I didn`t set joy[4] which by default is 0x00.      
When i take off, drone slide to the wall and i have no control over x and y direction. the only control i have is the z. 
Why would it happen?

zliDJI, 2017-09-13:
Could you try to pub the following topic in a terminal, in simulation environment?

rostopic pub -r 20 /dji_sdk/flight_control_setpoint_generic sensor_msgs/Joy "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
axes:
- 0.0
- 5.0
buttons:
- 0"

zliDJI, 2017-09-13:
Could you send the following command in a terminal in simulation?
```
rostopic pub -r 20 /dji_sdk/flight_control_setpoint_generic sensor_msgs/Joy "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
axes:
- 0.0
- 5.0
buttons:
- 0"

```

snakehaihai, 2017-09-14:
how to run dji simulation one? never tried. saw you guys have one in the dji assistant in the windows. does ubuntu have 1 as well?

zliDJI, 2017-09-14:
Simulation option is turned on using DJI Assistant 2 on Windows or Mac.

snakehaihai, 2017-09-14:
tested. works in simulation. will try with actual flight.

snakehaihai, 2017-09-14:
tried. in the simulation have the response. in actual no response. is my value given too small?

zliDJI, 2017-09-14:
1. You should take off before publishing to flight control topics.
2. You are using horizontal velocity, while in door there is no gps and so you don't have reliable velocity as feedback.

You want to verify if  the flight_control_...generic topic is working in real flight, I think a definitive way to verify that is to:
1. take your drone to a safe outside place, better tethered from below for safety.
2. write a small piece of code, such that, when you hit a button on your logitech joy, or flip the "H" switch on the LB2 RC, the code will send a constant small angle command at 20 Hz for 1 second and release sdk control or land.
3. take off and obtain control before you start your code in 2.

Thanks!

snakehaihai, 2017-09-14:
this drone can work indoor. no gps at all.
of you dont believe i can show you another video that we did. indoor position hold and navigation with 2X hokuyou lidar A3 on oldest 3.1 build.   The control was achieved by calling /drone->attitude_control(0x0a, y,x,z,yaw);, we have no problem in flying in 50mX300m fully covered metal container,  Thats why i was asking for similar control method as /drone->attitude_control with flag 0x0a

Problem is right now is I need to improve performance. say from +-5cm in position hold error. to +-1 cm in position hold error. To do that I need high fps IMU for vidual - inertial based navigation and lidar-inertial based navigation testing. previous 3.1 cant do this.

zliDJI, 2017-09-14:
uint8_t flag = (DJISDK::VERTICAL_VELOCITY |
DJISDK::HORIZONTAL_VELOCITY |  // --> change to DJISDK:: HORIZONTAL_ANGLE
DJISDK::YAW_RATE |
DJISDK::HORIZONTAL_BODY //|
//DJISDK::STABLE_ENABLE

snakehaihai, 2017-09-14:
k, will try it when i go to the lab. so as far as i can understand, a3 need the internal reference to execute the command, angle is using IMU. make sense. will get back to you later

snakehaihai, 2017-09-15:
tested it is woking
solution is -> change to DJISDK:: HORIZONTAL_ANGLE,
interesting to see why horizontal will affect vertical operation,
But there is still a problem of emergency by pass issue, switch to F mode while api is still obtaining control will cause both api and rc to lose control. how to solve that?

snakehaihai, 2017-09-18:
does DJISDK::STABLE_ENABLE required GPS? its IMU based or GPS based?

snakehaihai, 2017-09-18:
dji error insufficient payload..then uav go to full throttle and pitch back mode, F mode by pass not working, API by pass not working. how to solve this

snakehaihai, 2017-09-22:
add more payload. change mode to PAG then dont pub in main loop. can already

closed 2017-09-22 by snakehaihai