dji-sdk/Onboard-SDK-ROS #143: dji_sdk_node crashes, probably related to missions

tompe17, 2017-09-14:
I have problem with the dji_sdk_node crashing. It only happens maybe once per day so hard to debug. I managed to catch it in gdb but I do not know if the information is useful.  It crashed when running a sequence of missions.  Can I run out of available threads?

```
STATUS/1 @ getErrorCodeMessage, L590: droneTaskCallback
STATUS/1 @ getCMDIDTaskMSG, L762: CONTROLLER_FAIL
[ INFO] [1505397721.599130917]: initialized waypoint mission
[ INFO] [1505397722.641463712]: uploaded the 1th waypoint

[ INFO] [1505397723.681299405]: uploaded the 2th waypoint

[ INFO] [1505397724.681431841]: waypoint mission initialized and uploaded
[ INFO] [1505397822.379198566]: initialized waypoint mission
[ INFO] [1505397823.421117543]: uploaded the 1th waypoint

[ INFO] [1505397824.462409710]: uploaded the 2th waypoint

[ INFO] [1505397825.462531263]: waypoint mission initialized and uploaded
[ INFO] [1505397877.397428554]: initialized waypoint mission
[ INFO] [1505397878.439408686]: uploaded the 1th waypoint

[ INFO] [1505397879.479601508]: uploaded the 2th waypoint

[ INFO] [1505397880.479746852]: waypoint mission initialized and uploaded
[ INFO] [1505398051.294527777]: initialized waypoint mission
[ INFO] [1505398052.337478506]: uploaded the 1th waypoint

[ INFO] [1505398053.376588560]: uploaded the 2th waypoint

[ INFO] [1505398054.376738756]: waypoint mission initialized and uploaded
[ INFO] [1505398067.374491236]: initialized waypoint mission
[ INFO] [1505398068.415651295]: uploaded the 1th waypoint

[ INFO] [1505398069.456750706]: uploaded the 2th waypoint

[ INFO] [1505398070.456976049]: waypoint mission initialized and uploaded
[ INFO] [1505398094.773906691]: initialized waypoint mission
[ INFO] [1505398095.815682510]: uploaded the 1th waypoint

[ INFO] [1505398096.856624728]: uploaded the 2th waypoint

[ INFO] [1505398097.856915867]: waypoint mission initialized and uploaded

Thread 2 "dji_sdk_node" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff1a6f700 (LWP 19635)]
0x00007ffff7b67110 in ros::ServiceManager::lookupServicePublication(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/kinetic/lib/libroscpp.so

```
In GDB:

```
(gdb) bt
#0  0x00007ffff7b67110 in ros::ServiceManager::lookupServicePublication(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    () from /opt/ros/kinetic/lib/libroscpp.so
#1  0x00007ffff7b596b9 in ros::ServiceClientLink::handleHeader(ros::Header const&) () from /opt/ros/kinetic/lib/libroscpp.so
#2  0x00007ffff7b4315b in ros::ConnectionManager::onConnectionHeaderReceived(boost::shared_ptr<ros::Connection> const&, ros::Header const&) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#3  0x00007ffff7ae2f6b in ros::Connection::onHeaderRead(boost::shared_ptr<ros::Connection> const&, boost::shared_array<unsigned char> const&, unsigned int, bool) () from /opt/ros/kinetic/lib/libroscpp.so
#4  0x00007ffff7adf54b in ros::Connection::readTransport() ()
   from /opt/ros/kinetic/lib/libroscpp.so
#5  0x00007ffff7adfe14 in ros::Connection::read(unsigned int, boost::function<void (boost::shared_ptr<ros::Connection> const&, boost::shared_array<unsigned char> const&, unsigned int, bool)> const&) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#6  0x00007ffff7ae010f in ros::Connection::initialize(boost::shared_ptr<ros::Transport> const&, bool, boost::function<bool (boost::shared_ptr<ros::Connection> const&, ros::Header const&)> const&) () from /opt/ros/kinetic/lib/libroscpp.so
#7  0x00007ffff7b4559c in ros::ConnectionManager::tcprosAcceptConnection(boost::shared_ptr<ros::TransportTCP> const&) () from /opt/ros/kinetic/lib/libroscpp.so
#8  0x00007ffff7b57639 in ros::TransportTCP::socketUpdate(int) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#9  0x00007ffff7b8f8fb in ros::PollSet::update(int) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#10 0x00007ffff7b1a2f5 in ros::PollManager::threadFunc() ()
   from /opt/ros/kinetic/lib/libroscpp.so
#11 0x00007ffff5d225d5 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#12 0x00007ffff6fd06ba in start_thread (arg=0x7ffff1a6f700)
    at pthread_create.c:333
#13 0x00007ffff64653dd in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 

---------------------------

(gdb) info thread
  Id   Target Id         Frame 
  1    Thread 0x7ffff7fab780 (LWP 19266) "dji_sdk_node" 0x00007ffff6fd9c1d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
* 2    Thread 0x7ffff1a6f700 (LWP 19635) "dji_sdk_node" 0x00007ffff7b67110 in ros::ServiceManager::lookupServicePublication(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/kinetic/lib/libroscpp.so
  3    Thread 0x7ffff126e700 (LWP 19636) "dji_sdk_node" 0x00007ffff645b593 in select () at ../sysdeps/unix/syscall-template.S:84
  4    Thread 0x7ffff0a6d700 (LWP 19638) "dji_sdk_node" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  5    Thread 0x7fffebfff700 (LWP 19648) "dji_sdk_node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  6    Thread 0x7fffeb7fe700 (LWP 19793) "readPoll" 0x00007ffff6fd951d in read
    () at ../sysdeps/unix/syscall-template.S:84
  7    Thread 0x7fffeaffd700 (LWP 19795) "callback" 0x00007ffff642a30d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
  8    Thread 0x7fffea7fc700 (LWP 20864) "dji_sdk_node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  9    Thread 0x7fffe9ffb700 (LWP 20865) "dji_sdk_node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  10   Thread 0x7fffe97fa700 (LWP 20866) "dji_sdk_node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  11   Thread 0x7fffe8ff9700 (LWP 20867) "dji_sdk_node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  12   Thread 0x7fffcbfff700 (LWP 17507) "dji_sdk_node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225

```

tompe17, 2017-09-14:
ASynchSpinner ir used with 4 threads. I would guess that the crash is caused by two threads accessing the same data in some way. Is the locking handled properly?

Will the node work I it is changed to a single threaed spin?

tompe17, 2017-09-15:
I tested with ros::spin() instead of the ASynchSpinner but I still get crashes.

snakehaihai, 2017-09-17:
u need to provide a few more detailed description than this. whats pc u are running with. from the looks of it, seems like a memory issue which can be a bug or can be insufficient hardware

tompe17, 2017-09-17:
I am running on a NUC Core i5 or i7 (will check tomorrow) with a lot of memory. I have a test script now running my whole system which will produce the crash in maybe 15 minutes.  What I am basically doing is doing a lot of missions with two waypoints (the first waypoint is the current position). 

I will try to debug a bit more. Maybe run the node in valgrind. Any suggestion of things to test is welcome.

It looks like memory corruption of some sort.

snakehaihai, 2017-09-17:
try running in on a standard notebook or pc first?  the small embedded solution like nuc and tk1 tx1 are known to have those small issues. verified it works on small PC first? use 20 meter usb3 cable to fly the drone is small circle?

tompe17, 2017-09-18:
Well, a NUC is like a notebook without a screen. The error is actually triggered using the simulator so no need to fly it.

Have you uses missions? It only crashes when I use missions...

I am testing on a Matrice 100 if that matter.

snakehaihai, 2017-09-18:
Still NUC and notebook have some difference. I wasn`t running too much simulation since it wont help that much.
Lol I'm writing my own mission layer. My job is to do both indoor and outdoor navigation. GPS will always create the problem. So I implement my own mission control using mixed vision, lidar for autonomous guidance control(ie, point to point navigation with obstacle avoidance on). The only thing I call from sdk API is drone command yaw pitch roll and throttle.  Anything beyond that is my own implementation.  Minimize the risk of other ppl fucked in some memcp command.

tompe17, 2017-09-18:
Well, I have done both versions but thought I would save work and re-use the mission implementation since the control stuff is not our research focus.  I thought the software from dji would be well tested...

tompe17, 2017-09-18:
I got a crash in valgrind. The following looks suspicious:
```
==341== Thread 11:
==341== Invalid write of size 8
==341==    at 0x61AE82: DJI::OSDK::MissionManager::initWayptMission(int, void*) 
(dji_mission_manager.cpp:82)
==341==    by 0x61ACBF: DJI::OSDK::MissionManager::init(DJI::OSDK::DJI_MISSION_T
YPE, int, void*) (dji_mission_manager.cpp:43)

```

Soon after I got that I get a crash.

tompe17, 2017-09-18:
I added an assertion and it triggered:
```
[ INFO] [1505743800.585364283]: waypoint mission initialized and uploaded
[ INFO] [1505743823.244887720]: initialized waypoint mission
[ INFO] [1505743824.286097387]: uploaded the 1th waypoint

[ INFO] [1505743825.326199897]: uploaded the 2th waypoint

[ INFO] [1505743826.326347961]: waypoint mission initialized and uploaded
dji_sdk_node: /home/lrs/github/Onboard-SDK/osdk-core/api/src/dji_mission_manager.cpp:84: DJI::OSDK::ACK::ErrorCode DJI::OSDK::MissionManager::initWayptMission(int, DJI::OSDK::UserData): Assertion `wayptCounter < MAX_MISSION_SIZE' failed.

```

tompe17, 2017-09-18:
In dji_mission_manager.cpp there is the following code:
```
ACK::ErrorCode
MissionManager::initWayptMission(int timeout, UserData wayptData)
{

  wpMissionArray[wayptCounter] = new WaypointMission(this->vehicle);
  wpMission                    = wpMissionArray[wayptCounter];
  wayptCounter++;

  // @todo timeout needs to be defined somewhere globally
  return wpMission->init((WayPointInitSettings*)wayptData, timeout);
}

```

This code is very confusing. The wayptCounter is only increasing and I cannot find anywhere were it decreases. And it is really a waypoint mission counter I assume.

snakehaihai, 2017-09-18:
yes  you are right. I'm not running dji method so didn`t have this issue. I was using early apm script. which is neat and robust.
So for you, you might w8 for their support. Or you can also find some other ppl`s gps based navigation code. say px4 github space

tompe17, 2017-09-18:
In dji_mission_manager.hpp changing MAX_MISSION_SIZE from 5 to 5000 seems to solve the problem with crashes:
` static const int MAX_MISSION_SIZE = 5000;`
The code is very strange and I do not see how the current code for missions can work at all. To me it seems like after five missions the code will start to corrupt memory.  I also do not see why a history of missions are saved. It does not seem to be used anywhere. Unless it is so you can upload the next mission while the current missions is executing.

rohitsantdji, 2017-09-18:
@tompe17 The MissionManager abstraction was introduced more as a means of future-proofing (for example chaining a number of different missions and have them stream to the aircraft) rather than to have immediate use. Apologies if that makes the code harder to understand and less flexible than expected.

Could you help us understand your use case a little better, as to why your application requires more than five missions, each able to contain 99 waypoints, in a single OSDK session? We've kept the number low by default to avoid initializing a large memory space on machines that are considerably more memory constrained than a NUC, for example an STM32.

tompe17, 2017-09-18:
We have our ow mission language. That language has for example a flyto. I have implemented that flyto using a waypoint mission.

I do not understand what a OSDK session is. We start the system with the dji_sdk_node. Then we do a sequence of flyto, climb, scan (all using missions) without restarting anything.  So of course we will have more than 5 missions in 20-40 minutes or how long the battery lasts. 

The Waypoint Mission is used to implement other basic functionality. And the reason to use the mission for this is that it exists and we do not have to do the control plus more importantly it is displayed on the iPad what is being done. There is no API on the vehicle to display information in the iPad from what I have seen.

And even if you have an upper limit on the number of missions you should not corrupt the memory if this upper limit is exceeded. But to me it seems arbitrary and strange to have an upper limit at all.

tompe17, 2017-09-18:
Another reason to use mission is that we like the fact that we have some joystick control during the missions.

closed 2021-06-25 by dji-kyle