dji-sdk/Onboard-SDK-ROS #144: Python support question v2

jswoboda, 2017-09-14:
Referencing issue #110. I wasn't able to re open it so I just wanted to put my comment in a new issue so that will be tracked.

Hi, thanks for responding. I actually have had great success with your Python wrapper from ROS OSDK 3.2. It helped speed up my development and testing quite a bit. I understand after working with it that's it's just subscribing and calling services but this was my first time using ROS so it this was new for me.

There's a large interest in drones within the geosciences community for doing various kinds of measurements. Our community uses a lot of Python and having a wrapper can cut down on development time so we can focus on our science.

Sorry for the late reply. I'm currently sticking with v 3.2 for now. I'm using a Matrice 600 and I haven't updated the firmware to use v 3.3 with it. Can I get away with not doing this update? If I need to update the firmware can I still use drone.py?

lanyusea, 2017-09-15:
@jswoboda  could you specify which part does not work well with the update?

If any, I think it must be the best opportunity for a first-time ROS users to hands on and practice the ROS framework. Since the drone.py is the basicest ROS python implementation of topics/services/actions.

I'm looking forward to your and the geosciences community's pull request!

jswoboda, 2017-09-20:
Hi @lanyusea, I've just been reading that the newest firmware update for the Matrice 600 is compatible with 3.3 and not 3.2. The ROS OSDK for 3.3 does not have drone.py available currently within the repository. I would have to go to 3.2 to get the drone.py module and then place it in the 3.3 version.

jswoboda, 2017-10-04:
Hi @lanyusea, can I get an answer to the question about ros.py code working in with the new firmware update? I have not tested it with the new firware and sdk will there be an issue?

jswoboda, 2017-10-04:
Also did the messages for the global position data change? I've run my code in the simulator and I'm getting all zeros for the position data using the same calls as I did before. If I change out the message names in dji_drone.py will that correct the issue?

closed 2021-06-25 by dji-kyle