dji-sdk/Onboard-SDK-ROS #145: Unable to activate SDK on A3

zootboy, 2017-10-02:
I'm trying to use the ROS SDK to control my S900 with an A3 controller. I've gotten my developer API keys and added them to the launch file, but when I try to run I get a NEW_DEVICE_ERROR upon activation attempt:

> STATUS/1 @ init, L37: Attempting to open device /dev/ttyS1 with baudrate 230400...
> 
> STATUS/1 @ init, L47: ...Serial started successfully.
> [ INFO] [1506949425.497608986]: Succeeded to read from serial device
> 
> STATUS/1 @ init, L37: Attempting to open device /dev/ttyS1 with baudrate 230400...
> 
> STATUS/1 @ init, L47: ...Serial started successfully.
> 
> STATUS/1 @ parseDroneVersionInfo, L568: Device Serial No. = 0670125715
> 
> STATUS/1 @ parseDroneVersionInfo, L570: Hardware = A3
> 
> STATUS/1 @ parseDroneVersionInfo, L571: Firmware = 3.2.36.8
> 
> ERROR/1 @ initGimbal, L677: Failed to verify subscription!
> 
> ERROR/1 @ functionalSetUp, L157: Failed to initialize Gimbal!
> 
> ERROR/1 @ functionalSetUp, L187: Virtual RC not supported!
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ activate, L951: version 0x3022408
> 
> STATUS/1 @ getErrorCodeMessage, L590: activate
> STATUS/1 @ getCMDSetActivationMSG, L638: NEW_DEVICE_ERROR
> 
> ERROR/1 @ activate, L971: Failed to activate please retry SET 0x0 ID 0x1 code 0x3
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT
> 
> STATUS/1 @ getErrorCodeMessage, L590: removePackage
> STATUS/1 @ getCMDSetSubscribeMSG, L672: SUBSCRIBER_ILLEGAL_INPUT

The 3.2 FAQ mentions that this requires a Lightbridge 2 ( https://developer.dji.com/onboard-sdk/documentation/appendix/FAQ.html#a3-n3-do-i-need-a-lightbridge-2-rc-to-use-onboard-sdk ). I do not have a Lightbridge 2, nor do I plan on using one, and it would be pretty ridiculous to buy a $500 part to use once. Is there any way to activate the SDK without the Lightbridge 2? The computer that is running the SDK has internet access.

Thanks for any help!

rohitsantdji, 2017-10-02:
Hi @zootboy - the FAQ is out of date and will be updated soon. You don't need an LB2 anymore to activate - you can connect the A3 to a Windows/Mac computer running DJI Assistant 2, and as long as that computer has internet access the activation pipeline will route through Assistant instead of LB2. 

Please note that since this pipeline is still new and not as robust as the LB2 one, you might need to attempt activation a couple of times to get through to the servers.

zootboy, 2017-10-02:
Thanks a lot, this worked on the second try. I appreciate your help!

closed 2017-10-02 by zootboy

eduardofamendes, 2019-01-31:
What about the N3? I am also unable to activate my software.