dji-sdk/Onboard-SDK-ROS #148: dji_sdk/local_position how to calculate it from Ultra Wide Band sensor connected to Manifold using Matrice 100

trgiman, 2017-10-26:
Hi. We need to use MarvelMind UltraWideBand sensor (aka indoor GPS, similar results like with Vicon). I have found on dji dev forum, that we can change the On-board SDK ROS Package code in dji_sdk/local_position. DJI dji_sdk/local_position is currently calculated from GPS of matrice. So we need to calculate it from MarvelMind UltraWideBand sensor.

Marvel Mind Ultra Wide Band Supports UART or Virtual UART via USB or SPI. Could show us where to change such calculation?

rohitsantdji, 2017-10-27:
Hi @trgiman - the short answer is that this is not possible to do easily.

The long answer is something like this - it's possible, with a workflow that follows:

1. You will need to write a local position telemetry publisher that takes data from your UWB system and publishes to some topic with a message format identical to `dji_sdk/local_position`.
2. You will need to change the localPosition subscriber to listen to your topic instead of `dji_sdk/local_position`
3. Most importantly, you will need to write your own controller that uses position offsets and converts them to attitude+thrust commands, sent to the drone via publishing to the `/dji_sdk/flight_control_setpoint_generic` topic. (See the dji_sdk ROS [wiki](http://wiki.ros.org/dji_sdk) for more details on the message formats). You cannot use the position control topic that's provided as part of the SDK because the flight controller will not accept position commands in the absence of GPS, contrary to what the DJI forum post you linked seems to suggest.

I hope that helps answer your question.

trgiman, 2017-10-27:
Hi rohitsantdji,

Thanks for guidance. It is not that straightforward, exactly as you wrote. Could you please advice on 2 points below?

1.Our UWB device is sending NMEA 0183 format.I guess DJI Matrice 100 use uBlox chipset GPS. And that uBlox chipset is sending location updates in 10Hz (depends...)? We successfully connected our Marvel Mind UWB to PixHawk flight controllers and Naza flight controllers as well (just by setting right update frequency on UWB). So this would be the fastest way. Since N1 Matrice 100 flight controller would use MarvelMind UWB as GPS unit. And SDK would work as normal also for local positioning etc.But we do not know update frequency and also do not know if DJI Matrice 100 GPS sensor is sending directly NMEA 0183 format to the flight controller. This would save us a lot of hassle. Can somebody help about it? 

2. If option 1 is not reall really possible. Do you have got any sample code for building own controller position offsets converted into attitude+thrust commands? I have been looking on the ROS wiki. I do understand, that I need to insert additional flag to the messages, that have to be send to dji_sdk/flight_control_setpoint_generic. Still I am not sure how to build X Y Z world for Matrice from X Y Z received from UWB (in that case UWB does not have to send GPS NMEA 0183 fromat, it can send basically any X,Y,Z I guess). Sorry I am still new to On-board SDK.

Thank you.

rcabg, 2018-06-05:
Hello @trgiman ,

I'm struggling with the same thing trying to feedback the DJI autopilot with an external position system for indoor flying. Honestly, I'm trying to avoid solution no. 2. Did you solve this issue? It would be really really helpful to know your experience with this stuff. 

Thanks in advance!

trgiman, 2018-06-05:
hi @rcabg . I did not have any success so I moved to opensource pilots recently.

rcabg, 2018-06-05:
@trgiman thanks for you quick response! I will continue trying to make this work somehow. I will post if I get any progress.

IliasPap, 2018-06-21:
@trgiman  @rcabg  Hi I'm working with Ultra Wide Band sensors and I'm trying to do position hold of the drone calculating desired pitch and roll commands from position offset(PD control) but the drone is always doing oscillations

rcabg, 2018-06-21:
hey @IliasPap . Do Pozyx sensors give a not oscillating signal? If not, probably the PD parameters need to be tuned... can you describe your setup?

IliasPap, 2018-06-26:
I have connected a raspberry pi 3 with the drone and the sensor to raspberry pi. I calculate the position of the drone , publish it as a topic and then calculate the commands .The whole process takes 30-32 milliseconds , so the update frequency is approxiamately 30 Hz

rocklinsuv, 2018-07-20:
@IliasPap 
Hi IliasPap. So glad that I found someone encountered the same issue as me. Previously I used MarvelMind UWB to give my M100 position as well as velocity. When implementing the outer loop PID, I basically followed the traditional PID tuning guide, which was to increase Kp then add Ki and Kd. I found that no matter how large or small Kp was, the drone always oscillated and did not hold position. So did you manage to achieve position hold in the end? Any help would be very much appreciated.

Cheers,
rocklinsuv.

closed 2021-06-25 by dji-kyle