dji-sdk/Onboard-SDK-ROS #155: Drone Falls from sky when given waypoints

jswoboda, 2017-11-30:
I've left messages issues dji-sdk/Onboard-SDK#201 and dji-sdk/Onboard-SDK#216 and I was able to recreate the problem today. Basically under SDK control, I will send a list of waypoints to the drone and when the drone is starting its mission the drone starts dropping altitude. So far the operator has been able to take control before the drone hits the ground.

I've also noticed that the altitudes read from the gps coordinates are negative. The altitude coordinates that I'm given are approximately -50 m when it was about 3 m in the air with ground sea level at 82 m. Also the height reading is only zero. The simulations for this work perfectly and I'm getting information out of the height subscriber as well. The height measurements on the DJI Go app that our operator uses seems accurate though.

I have been using a Matrice 600 pro, and seen this behavior with API version 3.3 and 3.4. I have logs from both ros and the drone and can make them available, I just need to know how I need to make them available to the developers. 

As I've stated before, **THIS IS A MAJOR SAFETY ISSUE!** I would like to get this fixed as soon as possible.

eeroniemi, 2017-12-01:
I have also noticed that altitude number from ROS topic `/dji_sdk/gps_position` don't make any sense, but is quite random (using M100 here). Setting altitude number to use either fusioned or RAW data didn't change anything - altitude numbers were still random.

I asked DJI support about this and got this answer: "The height value needs to be filtered." Not sure exactly what this means but our workaround is to call ROS service `/dji_sdk/set_local_pos_ref` (after we get good GPS signal) and read altitude from topic `/dji_sdk/local_position`. This gives us altitude relative to the takeoff point's ground level, so you get 0 altitude for your takeoff point.
 
If you look at the Ground Station Protocol documentation (https://developer.dji.com/onboard-sdk/documentation/introduction/ground-station-protocol.html) it says that all altitude parameters are "relative altitude from takeoff point". You need to take this into account when uploading waypoint missions (don't use absolute GPS altitude numbers on certain lat/lon point, but height relative to takeoff point).

amenonDJI, 2018-01-24:
@eeroniemi Please reference [this](https://github.com/dji-sdk/Onboard-SDK/wiki/Altitude-Reference-for-different-topics) wiki page on more information regarding Altitude reference

jswoboda, 2018-04-24:
Hi, 

This issue still persists the drone will not properly fly to waypoints and I'm getting 0s from the height above take off subscription, basically the same behavior as in dji-sdk/Onboard-SDK#218.  I have done multiple tests with both under computer control and remote pilot control and the height is still 0 at all times. I've set the ground reference, both with a service call before take off and setting the arm command. 

The other issue is that this issue is not present in the simulator! In the simulator the read out. For example here are a log read out for the lat long altitude and height above takeoff 

> 2018-04-24 14:52:33,722 - dronelog - INFO - Distance 0.386677265167
2018-04-24 14:52:33,724 - dronelog - INFO - lat 22.542813, long 113.958902, alt 4.61332273483, height 4.61332273483
2018-04-24 14:52:33,724 - dronelog - INFO - x -5.69818837143e-08, y -1.81132715168e-07, z 4.61332273483
2018-04-24 14:52:33,725 - dronelog - INFO - Power 50.0

When the drone is flying and hovering above in a real flight:

> 2018-04-24 15:51:05,936 - dronelog - INFO - Distance 5.14500734476
2018-04-24 15:51:05,938 - dronelog - INFO - lat 42.6098032189, long -71.4944498202, alt -22.9147491455, height -0.0
2018-04-24 15:51:05,938 - dronelog - INFO - x 0.370596108594, y 0.409285747799, z 3.47104263306
2018-04-24 15:51:05,939 - dronelog - INFO - Power 90.0

The local position height makes sense from what we saw on the DJI Go app. The GPS height is problematic because of the ellipsoidal height translation issues referenced in the wiki page. This was also referenced in other issues, dji-sdk/Onboard-SDK#158, dji-sdk/Onboard-SDK#218 and others.

tompe17, 2018-04-30:
The problem with height_above_takeoff being zero is still not fixed.  This caused us to have issues with commanding the M600 to fly to 0 height.  Fortunately the backup pilot could takeover.  But this is not good since it worked in the simulator and we actually trusted that the simulator would behave as the real vehicle in most respects.

jswoboda, 2018-05-10:
Hi @tompe17, have you had any luck getting a resolution to this problem?

tompe17, 2018-05-15:
Not really. I implemented a ROS server that keeps track of the GPS Altitude when doing take-off and using that to compute the height argument needed for service calls. It works but it do not fell as robust as it should be...

InterestingWalrus, 2018-12-12:
Hi @jswoboda, this issue happened to me today albeit with a M100. Did you ever come across a workaround for this?

jswoboda, 2018-12-19:
Nope, I gave up on DJI flight controller. I've moved to pixhawk flight controllers.

closed 2021-06-25 by dji-kyle