dji-sdk/Onboard-SDK-ROS #158: [Version 3.4] High noise on local_pose/Z value

prodrone-dev, 2017-12-20:
Hello there!

We were just testing the performance of update 3.4 over 3.3 and noticed significant increase in noise of altitude estimation in topic `/dji_sdk/local_position`.

Here is the comparison of flying SDK 3.3 to 3.4. 
- Controller A3 on S1000 frame
- 3.3 was flown in more windy day with some gusts
- 3.4 has zero wind flight log

Version 3.3 flight:
[image]

Version 3.4 flight:
[image]

You can clearly see how 3.4 altitude jumps within almost 1 meter to compare with 3.3 smooth estimation. It's not a problem for us to implement some LPF filter over the altitude, but we would rather prefer you doing that on more low level, so that if one day you decide to improve it (and we miss it), we wouldn't have double filters on top of the system.

Thank you in advance,
ProDrone team.

JimTheKiwi, 2017-12-20:
This is interesting data... can you determine which was closer to the reality of the drone altitude?  IE:
 - On a gusty day, did 3.3 record a smoother flight than the drone actually made (removing real detail before saving the recording)? 
 - On a smooth day, did 3.4 record a rougher ride than the drone actually made (adding noisy detail that did not actually happen)?
 - A mixture somewhere in between?

Ideally the recorded data should be as close as possible to reality, whether that was rough or smooth.

The rough rotation of the drone is normally mostly cancelled out by the rotation of the gimbal to give the camera a more steady pose.  If you artificially smooth the drone movement more than the gimbal movement then the combined tracking drone+gimbal=camera would show rougher movement than the camera actually experienced.

I saw in SDK-ROS 3.2.2 some symptoms that I suspect may have been due to different smoothing for the (M600) local_position compared to the (Ronin-MX) gimbal pose.  I will be retesting this in early 2018 using SDK 3.5+ with the latest firmware and using the hardware timing sync pin (introduced in the A3/M600 firmware upgrade that came with OSDK 3.3).

Thanks!

prodrone-dev, 2017-12-20:
@JimTheKiwi 
I am really sorry but comparison was made between versions 3.2.2 and 3.4. I've already updated the initial issue.

Responding to your questions:

> On a gusty day, did 3.3 record a smoother flight than the drone actually made (removing real detail before saving the recording)?

On a gusty day 3.2.2 was showing similar altitude hold as on the log. You can observe around 0.5 meter altitude drop between 15-45 sec, I would say visually it dropped 1 meter.

> On a smooth day, did 3.4 record a rougher ride than the drone actually made (adding noisy detail that did not actually happen)?

Exactly! The drone was very solid on altitude hold, I would even say perfect hold, nothing even close to what is on the logs.

> A mixture somewhere in between?

I would say the 3.2.2 performance is much closer to what is observed in reality.

ChrelleP, 2017-12-29:
We are experiencing the same issues, and are very interested in finding a solution.

justwillim, 2017-12-29:
in 3.4 we changed to use raw data instead fusion-ed data:
3.4:
'''
    if(local_pos_ref_set)
    {
      geometry_msgs::PointStamped local_pos;
      local_pos.header.frame_id = "/local";
      local_pos.header.stamp = gps_pos.header.stamp;
      gpsConvertENU(local_pos.point.x, local_pos.point.y, gps_pos.longitude,
                       gps_pos.latitude, this->local_pos_ref_longitude, this->local_pos_ref_latitude);
      local_pos.point.z = gps_pos.altitude - this->local_pos_ref_altitude;
      /*!
      * note: We are now following REP 103 to use ENU for
      *       short-range Cartesian representations. Local position is published
      *       in ENU Frame
      */

      this->local_position_publisher.publish(local_pos);
    }
'''
3.2:
'''
        local_position.header.frame_id = "/world";
        local_position.header.stamp = current_time;
        gps_convert_ned(
                local_position.x,
                local_position.y,
                global_position.longitude,
                global_position.latitude,
                global_position_ref.longitude,
                global_position_ref.latitude
                );
        local_position.z = global_position.height;
        local_position.ts = global_position.ts;
        local_position_ref = local_position;
        local_position_publisher.publish(local_position);
'''

JimTheKiwi, 2017-12-29:
Thank you @justwillim for showing us the next layer, we need to go a few more layers deeper to understand what our options are.

You showed us code in branch 3.4 & 3.5 inside [DJISDKNode::dataBroadcastCallback()](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.5/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L107), and looking a few lines earlier it looks like the data came from the new A3 firmware:
```
    DJI::OSDK::Telemetry::GlobalPosition global_pos =
	      vehicle->broadcast->getGlobalPosition();
    ...
    gps_pos.altitude = global_pos.altitude;
    ...
    local_pos.point.z = gps_pos.altitude - this->local_pos_ref_altitude;
    this->local_position_publisher.publish(local_pos);

```
However, there is almost identical code inside [DJISDKNode::publish50HzData()](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.5/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L322) which uses different A3 telemetry data apparently fusing GPS and IMU:
```
  Telemetry::TypeMap<Telemetry::TOPIC_GPS_FUSED>::type fused_gps =
	    vehicle->subscribe->getValue<Telemetry::TOPIC_GPS_FUSED>();
    ...
    gps_pos.altitude = fused_gps.altitude; 
    ...
    local_pos.point.z = gps_pos.altitude - p->local_pos_ref_altitude;
    p->local_position_publisher.publish(local_pos);

```
In [dji_telemetry.hpp](https://github.com/dji-sdk/Onboard-SDK/blob/3.5/osdk-core/api/inc/dji_telemetry.hpp#L238) I see definitions for GlobalPosition and for GPSFused with comments that it is GPS fused with IMU.

Is the local_position data being published by DJISDKNode::dataBroadcastCallback() from vehicle->broadcast->getGlobalPosition() or by DJISDKNode::publish50HzData() from vehicle->subscribe->getValue<Telemetry::TOPIC_GPS_FUSED>()?

If **BOTH** are publishing, would that perhaps keep the local_position values jumping back and forth between the two different A3 telemetry sources, less stable than either source alone?  The graph from @prodrone-dev does sometimes appear to be jumping between two more stable levels, I can’t tell.

rohitsantdji, 2017-12-29:
@JimTheKiwi @prodrone-dev I think there seems to be some amount of confusion regarding the sources of altitude data. 

The two functions you linked to -  `DJISDKNode::dataBroadcastCallback()` and `DJISDKNode::publish50HzData()` - are mutually exclusive in operation. They are using the *Broadcast* and *Subscription* styles of getting telemetry data respectively. The differences between the two mechanisms are outlined [here](https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-telemetry.html). You can pick between either using a parameter in the sdk.launch file.
[image]

As @justwillim pointed out, in SDK 3.2 we were publishing fused Broadcast telemetry data, specifically the `height` parameter, to the local_position topic. In SDK 3.3+ with Subscription telemetry, the flight controller outputs four data streams from which you can obtain altitude:
```
TOPIC_ALTITUDE_FUSIONED
TOPIC_ALTITUDE_BAROMETER
TOPIC_GPS_FUSED
TOPIC_GPS_POSITION
```

The short answer for the noise levels is that TOPIC_ALTITUDE_FUSIONED is equivalent to the fused data that we used to obtain through Broadcast in SDK 3.2, while TOPIC_GPS_FUSED is the gps/baro fusion data that we are publishing in SDK 3.3+; this does not include IMU as well as smoothing filters and results in the high noise you are seeing now. TOPIC_ALTITUDE_BAROMETER is almost equivalent to the altitude in TOPIC_GPS_FUSED, and TOPIC_GPS_POSITION outputs raw GPS altitude.

We are currently working on doing experiments to verify the altitude references and characterize the variations among the various sources. We are also working on some documentation that will help clarify this issue a little more, as well as shedding light on the various height/altitude issues reported in #161, dji-sdk/Onboard-SDK#201, dji-sdk/Onboard-SDK#216, dji-sdk/Onboard-SDK#218. Please bear with us a little longer!

amenonDJI, 2018-01-24:
Please reference [this](https://github.com/dji-sdk/Onboard-SDK/wiki/Altitude-Reference-for-different-topics) wiki page for more information regarding Altitude reference

closed 2021-06-25 by dji-kyle