dji-sdk/Onboard-SDK-ROS #162: Flight Control Sample Bug

amenonDJI, 2018-01-04:
A bug was seen in Flight control sample where the aircraft loses altitude even when positive Z offset commands were given. See issue #201 for reference. We are investigating the issue and we are looking to have a fix for the next release. Until then, please do not use Flight Control sample in the Linux Onboard SDK and ROS versions.

prodrone-dev, 2018-01-04:
@amenonDJI Could please elaborate on this one. 

Is it safe to use current ROS-SDK v3.4? We have several drones flying in production with the latest DJI-ROS-SDK (3.4) and we would like to know if this issue might cause the crash.

Is it a problem only for [Onboard-SDK sample code](https://github.com/dxywing1314/Onboard-SDK/blob/3.3/sample/linux/flight-control/flight_control_sample.cpp) and [ROS-SDK demos](https://github.com/dji-sdk/Onboard-SDK-ROS/tree/3.5/dji_sdk_demo/src) or it can effect any flight code made by 3rd party developers?

Kindly ask you for quick reply,
thank you,
ProDrone dev team.

amenonDJI, 2018-01-04:
@prodrone-dev The OSDK core and the ROS main node are not affected. Only the flight control sample is affected. 

The issue with the sample is that we are referencing the Aircraft Altitude instead of Aircraft height. When a Z offset command is given, the current code commands the aircraft with respect to the Altitude, when it should be referencing the Height. We are still in the process of investigating the issue.

stotime123, 2018-02-14:
Hey all, I was wondering if there were any updates on this? We just got a Matrice 100 and same issue pops up. Is the onboard SDK v3.6 compatible with Matrice 100?

rohitsantdji, 2018-02-14:
OSDK 3.6 fixes this issue for all aircraft, but has a small bug that prevents M100 from working correctly. This bug will be fixed tomorrow, so you should be good to go after that.

stotime123, 2018-02-14:
Thanks Rohit -- any idea when the fix is going to be released today?

Also I am guessing this is what you mean by "bug preventing M100 working correctly for 3.6"
"ACK_NO_RESPONSE_ERROR" on initVehicle ?

closed 2021-06-25 by dji-kyle