dji-sdk/Onboard-SDK-ROS #164: [Version 3.5] Serial Device Validation error with M210

soutieslak, 2018-01-10:
I am running the Onboard-SDK-ROS 3.5 package together with the Matrice M210. I get the following error after the serial connection is started: "_[ERROR] [1515584789.178233011]: Failed to read from serial device. The Onboard SDK is not communicating with your drone.
[ERROR] [1515584789.178682103]: Vehicle initialization failed_" Please see the screenshot below:
[image]

I found a similar issue that a user had with the M600 running the Onboard-SDK 3.3 over here:
[](https://github.com/dji-sdk/Onboard-SDK/issues/176)

He commented out the serial status in the _dji_sdk_node.cpp_ file, see image below:
[image]

I did the same and it allowed ROS to run. I could then subscribe to various topics without any problem.

rohitsantdji, 2018-01-10:
Could you tell us what onboard computer are you using?

Also, for correct operation of OSDK-ROS, we recommend a baud rate of 921600 (see image below taken from wiki.ros.org/dji_sdk). Without commenting those lines in dji_sdk_node, does your setup work at that baud rate?
[image]

soutieslak, 2018-01-11:
Hi, thank you for pointing that out! I did not see that the baud rate should be set at 921600 for the N3. My drone is currently set to the DJI Assistant 2 default value of 230400, therefore I also set my ROS launch file to that value. 

I will change the drones baud rate to 921600 and then also change the launch file and see if that lets the above setup status check pass. Will report back later. 

Thanks again for your input, appreciate the reply.

soutieslak, 2018-01-11:
Okay I changed both the drones and ROS baud rates to 921600 and still get the same error. See screenshot below:

[image]

Still going to stick to commenting out that checkpoint in the code unless someone might have a better suggestion?

snakehaihai, 2018-01-16:
we successfully run M210 with 3.4. try swap tx rx. we made a mixtake there oso

soutieslak, 2018-01-16:
@snakehaihai thanks for the reply, yeah I tried that, but its not the issue. As everything works when just comenting out those 10 lines of code above. All data does come through therefore the Tx and Rx must be correct. Not sure what is raising that flag.

Have you guys tried 3.5 yet?

lovesmall, 2018-01-16:
Now my M210 is fly with 3.4. I tried with 3.5 oso. works. as long as your firmware version is same requirement and tx rx are ok. should have no problem. OSO do remove usb kernel deplay and give access right. 
BTW this is my friend account in other PC

Ebtiisam, 2019-10-15:
How do you swap tx and rx? 
what is 3.4oso ?
also baud rate for m210 and manifold 2 c is ? because i'm using both and m210 is 921600 while manifold 2c is 115200

closed 2021-06-25 by dji-kyle