dji-sdk/Onboard-SDK-ROS #167: RTK_* TOPICs Not Published on ROS Mobile SDK 3.5 on A3 + RTK

almapa, 2018-01-18:
we have the following setup:

Onboard SDK 3.5, A3  + RTK, Latest FW,

we see that RTK TOPICS are not published, in particular callback "publish10HzData" is called (sometime) only once. Therefore also BATTERY_INFO is not published. 

in case  RTK_* topics are moved to 50Hz pusblishing function, then /dji_sdk/battery_state topic is correctly published while the others 50Hz topics (like as /dji_sdk/gps_health) are not anymore published.

Could it be some problem regarding RTK_* topics management in Onboard SDK library?

Have someone already tested RTK on A3, by Onboard SDK and ROS?

thanks

rohitsantdji, 2018-01-18:
Are you running the OSDK at 921600 baud rate? If you run at a lower rate you might run into issues of data dropout.

almapa, 2018-01-20:
effectively incrementing serial line baud rate rtk topics are published now
thanks