dji-sdk/Onboard-SDK-ROS #169: [M600][demo_flight_control.cpp] Distance to home_position to can flight

dangviethieu-hntv, 2018-01-22:
Hi , I run demo_flight_control.cpp in Simulator and had issue. If Distance from end_position to home_position (where to takeoff) greater than 20m, Drone will fly to points has distance to home_position 18-20m and stop there. I run many time and this always give results same.
Please confirm with me min distance to home_position to drone can fly.
Thanks very much!

rohitsantdji, 2018-01-22:
You will need to run DJI Go to an iOS/Android device connected to the RC, and monitor the app for any no-fly zone warnings or similar dialogs that might pop up. Do let us know if that doesn't solve your issue.

dangviethieu-hntv, 2018-01-23:
Yes, I turned on RC and run DJI Go, run code demo. But if I change setTarget(0, 20, 3, 60) to setTarget(0, 25, 3, 60) **(>20m)** , drone will fly to points has distance to home_point is 18-20m and stop there.
`if(takeoff_result)`
`{`   
    `square_mission.reset();`
    `square_mission.start_gps_location = current_gps;`
    `square_mission.setTarget(0, 20, 3, 60);`
    `square_mission.state = 1;`
    `ROS_INFO("##### Start route %d ....", square_mission.state);`
 `}`
So distance from end_point to home_point must less than 20m ??? 
Please help me confirm it.

rohitsantdji, 2018-01-23:
This is not expected behavior. This might happen due to the following reasons:

1. The simulated co-ordinates are in Mainland China, and your DJI account is not configured for that
2. The aircraft is in Beginner mode
3. Your simulated co-ordinates are near a no-fly zone.

If you aren't receiving any no-fly zone warnings, you might want to debug for the first two possibilities. The beginner mode toggle can be found in DJI Go in the menu on the top right corner. Also, could you tell us your simulated co-ordinates?

dangviethieu-hntv, 2018-01-24:
Yes, I checked and detected my aircraft is in Beginner mode. Then I change it and set distance limit is 500m and my drone flied very nice. 
So Thank you very very much about helpful. Wonderful !!!

closed 2021-06-25 by dji-kyle