dji-sdk/Onboard-SDK-ROS #170: Time is missing in RTK-topic

Dev-Rune, 2018-01-23:
**Line 267 in file "dji_sdk_node_publisher.cpp".** 
The header (time-stamp and frame id) in the rtk_position message is not set. Please add:
```
  rtk_position.header.frame_id = "/gps";
  rtk_position.header.stamp    = msg_time;
```

Without these lines the header is empty, which makes it difficult to determine when the specific RTK-GPS position was measured.

aybakana, 2019-03-12:
Did you resolve your problem? which version do you use? I cant even see RTK topics.

Dev-Rune, 2019-03-12:
I forked the code (v. 3.3 i believe) and added the frame and timestamp to the file myself . I also made a pull request, but I had no response from the DJI-team, regarding this issue. They may have corrected this issue in newer versions.

closed 2021-06-25 by dji-kyle