dji-sdk/Onboard-SDK-ROS #174: Onboard cannot reconnect to drone after ROS node force quit

zootboy, 2018-02-07:
Running the 3.6 branches of the onboard SDK and the onboard ROS node. My hardware is an A3 running firmware 1.7.1.5.

When launching the onboard SDK node, it connects to the drone fine. If it exit the node by doing SIGINT and let it exit cleanly, it reconnects to the drone fine on subsequent launches. However if the node exits uncleanly (e.g. kill -9 on the node process), subsequent launches of the node fail with the following error:

```
ROS_MASTER_URI=http://127.0.0.1:11311/

process[dji_sdk-1]: started with pid [11436]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyS1 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.
[ INFO] [1518067250.566967141]: Succeeded to read from serial device

STATUS/1 @ init, L55: Attempting to open device /dev/ttyS1 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1332: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are inverted.
		* Serial port is occupied by another program.
		* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

[ERROR] [1518067256.572954100]: drone activation error
[ERROR] [1518067256.573441275]: Vehicle initialization failed
```

The only way to recover is to power cycle the drone controller.

zhangkun35268, 2018-03-19:
I have the same problem， Use PL2303 to connect A3 and raspberry pi3.

User Configuration read successfully.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1339: Drone version not obtained! Please do not proceed.
Possible reasons:
        Serial port connection:
                * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
                * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
                * TX and RX pins are inverted.
                * Serial port is occupied by another program.
                * Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1228: Unable to initialize some vehicle components!
STATUS/1 @ getErrorCodeMessage, L644: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L710: ACK_NO_RESPONSE_ERROR
Vehicle not initialized, exiting.

HuangDihe, 2021-01-15:
I got the  same  problem  with  A3.    But  I   got  a  lot   of  data  in the serial usb.

closed 2021-06-25 by dji-kyle