dji-sdk/Onboard-SDK-ROS #176: [Version 3.6] Activation fails on Ubuntu 16

prodrone-dev, 2018-02-08:
Dear DJI team,

we have recently changed the version of Ubuntu to 16.04 from 14.04 and we started to encounter an issue with obtaining drone version via ROS SDK.

Hardware:  `Odroid XU4`,  `A3 controller`
Sofware: `Ubuntu 16.04` , `ROS Kinect`, `DJI ROS SDK` `3.6.0` -  `3.4.0`
Tested with these kernel versions:
```
Linux odroid 4.9.44-56 #1 SMP PREEMPT Thu Aug 24 10:38:35 UTC 2017 armv7l armv7l armv7l GNU/Linux
Linux odroid 4.14.5-92 #1 SMP PREEMPT Mon Dec 11 15:48:15 UTC 2017 armv7l armv7l armv7l GNU/Linux
```

The connection fails every time we run sdk.launch file under Ubuntu 16.04. The issue persist between DJI ROS SDK versions 3.6 and 3.4.
The issue as well persist even with updated Linux Kernel. These versions were tested:
`ubuntu-16.04.3-4.14-minimal-odroid-xu4-20171213`
`ubuntu-16.04.3-4.9-minimal-odroid-xu4-20170824`
Source:  https://odroid.in/?directory=.%2Fubuntu_16.04lts%2F

Same setup, running Ubuntu 14.04 works perfectly fine. So hardware issues are ruled out.

All these possible reasons to fail were checked and resolved:
```
Serial port connection:
        * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
        * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
        * TX and RX pins are inverted.
        * Serial port is occupied by another program.
        * Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again
```

Here is the terminal output of running the sdk.launch and the Port Setup and the list of tested Ubuntu versions :
```
root@odroid:~# roslaunch dji_sdk sdk.launch 
... logging to /root/.ros/log/c303950e-0ce0-11e8-814c-001e06306fd2/roslaunch-odroid-1300.log 
Checking log directory for disk usage. This may take awhile. 
Press Ctrl-C to interrupt 
Done checking log file disk usage. Usage is <1GB. 

started roslaunch server http://odroid:40041/ 

SUMMARY 
======== 

PARAMETERS 
 * /dji_sdk/align_time: False 
 * /dji_sdk/app_id: ---------REMOVED----------
 * /dji_sdk/app_version: 1 
 * /dji_sdk/baud_rate: 921600 
 * /dji_sdk/enc_key: ---------REMOVED----------
 * /dji_sdk/serial_name: /dev/ttySAC0 
 * /dji_sdk/use_broadcast: False 
 * /rosdistro: kinetic 
 * /rosversion: 1.12.12 

NODES 
  / 
    dji_sdk (dji_sdk/dji_sdk_node) 

auto-starting new master 
process[master]: started with pid [1310] 
ROS_MASTER_URI=http://localhost:11311 

setting /run_id to c303950e-0ce0-11e8-814c-001e06306fd2 
process[rosout-1]: started with pid [1323] 
started core service [/rosout] 
process[dji_sdk-2]: started with pid [1326] 

STATUS/1 @ init, L37: Attempting to open device /dev/ttySAC0 with baudrate 921600... 
 
STATUS/1 @ init, L47: ...Serial started successfully. 
[ INFO] [1518102013.423673515]: Succeeded to read from serial device 

STATUS/1 @ init, L37: Attempting to open device /dev/ttySAC0 with baudrate 921600... 

STATUS/1 @ init, L47: ...Serial started successfully. 

ERROR/1 @ parseDroneVersionInfo, L449: Drone version was not obtained. Please restart the program or call getDroneVersion 

ERROR/1 @ getDroneVersion, L1142: Drone version not obtained! Please do not proceed. 
Check if your serial connection is okay, and if your activation status is okay. 

ERROR/1 @ functionalSetUp, L121: Failed to initialize Version! Please exit. 
[ERROR] [1518102020.425302156]: Vehicle initialization failed 
^C[dji_sdk-2] killing on exit 
[rosout-1] killing on exit 
[master] killing on exit 
shutting down processing monitor... 
... shutting down processing monitor complete 
done 
```

Information about UART0 Port:
```
root@odroid:~# sudo stty -F /dev/ttySAC0 -a 
speed 921600 baud; rows 0; columns 0; line = 0; 
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>; eol = <undef>; eol2 = <undef>; 
swtch = <undef>; start = <undef>; stop = <undef>; susp = <undef>; rprnt = <undef>; werase = <undef>; lnext = <undef>; 
discard = <undef>; min = 0; time = 0; 
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread clocal -crtscts 
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8 
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0 
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc 
```

Thank you in advance,
ProDrone dev team.

dunk-emesent, 2018-02-08:
Try catting the serial port. i.e.

    $ cat /dev/ttySAC0

You should receive something (it'll appear as garbage on the terminal). If not, you have a serial port issue.

prodrone-dev, 2018-02-09:
@dunk-emesent we get the output on serial port and it different before and after `sdk.launch` starts.

Here is the output before the launch:
```
20fe 0242 f830 ff20 10ff 31ff 0200 f8b0
fe00 ff00 fe20 ff00 02e0 30ff b0fe b1ff
32fe 0040 f830 ff22 fe30 ffb1 ff00 f831
ffb0 ff20 fe30 fe00 00f0 30ff 30ff 20ff
30ff 0000 f0b0 ff20 ff20 fe30 ff00 00f0
32ff 00fe 30ff 00ff 0000 f020 feb0 fe22
fe30 fe00 81f0 30ff 40ff b0ff 20fe 0000
f0b1 feb0 fe32 fe30 ff00 00fc 20ff 30fe
30ff 00fe 0000 f420 feb1 ff20 fe30 ff00
01f8 40ff 42ff 32fe 30ff 00fe b0fe 32ff
b0ff 31fe 0008 f800 ff30 fe32 ff20 ff02
00f8 00fe 30fe 00ff 30ff 0010 fe30 ff32
ff02 ff32 ff00 01f8 30ff b0fe 20fe 30ff
0080 f020 fe30 fe20 fe30 fe00 08f8 b1ff
2030 fe20 ff02 01f8 20ff 10ff 20fe 02fe
1000 f831 ff20 fe20 feb0 fe00 02f8 32fe
40ff 00fe b0ff 0000 f832 fe00 ff32 ff60
1000 f820 fe32 ff30 ffb1 ff02 00f0 32fe
30fe 00fe 00ff 0208 f000 ff30 ff20 feb1
fe01 00fc 30ff ff30 fe32 fe00 00f8 30fe
b0ff 40ff 20fe 0000 fc30 fe22 20fe 32fe
0201 f830 feb1 ffb1 fe30 fe02 08e0 b1ff
30fe 20fe 20fe 0041 f820 ff00 fffe 30fe
0010 fe20 feb0 ff42 ff32 ff00 10fe 30fe
32ff 20ff 80fe 0220 fe20 00ff 2030 ff00
08f8 32ff 02fe 00ff 00fe 00fe 00ff a0ff
32fe 30fe 0008 f820 ff20 ff30 ff32 fe00
81f8 31ff 30fe 02ff 20fe 0041 f830 ff32
feb0 fe20 ff00 00e0 20fe 00ff b1fe 22ff
0001 f800 fe20 fe30 fe30 fe02 91f0 30ff
30fe 30fe 30ff 0200 f830 feb0 ff32 feb1
ff00 80f0 20fe b0ff 00ff 30fe 0000 f830
fe60 32ff 30ff 0000 f822 fe30 fe30 ff20
fe02 00f8 22fe 20fe 00fe 30ff 00f8 20fe
31fe 30ff 6000 80f0 20fe 20ff b0ff 30fe
0000 f402 ff00 fe22 32fe 0210 fe32 ff31
ff00 ff30 fe02 00f8 32ff 30ff b1ff b1ff
00f8 30ff b0fe 31fe 20ff 0100 f800 ffb0
fe42 ff20 fe00 00f8 32ff 2022 ffb0 fe00
fc20 feb0 ff60 32ff 0041 f820 ff30 fe00
fe30 fe01 00fc 32fe 30ff 00ff 30ff 0241
f840 ff20 feb0 ff20 fe10 02f4 40ff 20fe
30fe 31fe 0100 fc30 ff30 ff30 ff22 0000
fc32 feb0 fe32 fe00 fe01 00fc 00fe 20fe
ff02 fe00 01f8 b0ff 32fe b1ff 32ff 0083
f020 fe32 fe30 fe20 0008 f0b1 feff 30ff
30ff 0001 f840 ff30 ff00 fe40 0001 f830
ff00 feb0 fe20 ff00 01f8 b0ff 30fe 30fe
ff00 00f0 32fe 20ff 00fe b1fe 0000 f820
ff30 fe32 ff00 fe00 44f8 32fe 00fe 80ff
02fe 0000 fc32 ff30 ff60 b0fe 0001 f820
ffa0 fe30 ff22 fe00 ff20 fe20 fe30 feb0
ff02 49f8 b1fe 20ff 20fe 32ff 0002 ff30
fe30 ff30 fe20 ff01 02fc 32fe 30fe 30ff
20ff 0008 f002 feb1 fe30 fe30 fe00 41e8
30fe 30fe b1ff 20fe 00fc 42ff 32fe 30ff
20fe 0010 fe30 ff20 feb1 ffb0 ff00 00fc
20ff b1fe 20fe b0fe 0210 fe22 ff20 fe10
fe40 ff00 01f8 32ff 30ff 30fe 00ff 0000
fc00 ff20 feb1 ff22 0205 fc00 fffe 32ff
20ff 00f8 30ff 30fe 20fe 2000 00f0 40ff
b1ff 30ff 30fe 0000 f030 ffb1 feb1 ff30
fe00 80f8 b1fe 30fe 20fe 32ff 0000 fc00
ff20 fe20 fe60 0200 f822 30fe 10fe b1ff
0010 fe00 ff32 ff22 ff32 fe00 00fc 02ff
30ff 20ff 30fe 0101 f820 fffe 20ff 20ff
0200 f020 ff20 fe30 ffb1 ff02 00fc 32ff
fe42 ff30 ff00 00f0 b0fe 30ff 30fe 32ff
0201 f0b0 fe20 ff22 b2ff 0008 f830 ff30
ff30 fe32 fe02 01f8 30fe 4200 fe00 fe02
42f8 32ff 02fe 00fe 22fe 0100 f820 fe30
fe30 ff20 fe00 83f0 00fe 2200 ff20 fe00
00f8 20fe 20ff 20fe 40ff 1000 f830 fe30
feff 30fe 0000 f020 feb1 ffb1 fe30 ff00
01f8 30ff 82fe 30ff 2000 08e0 40ff 20fe
40ff b1fe 0000 fc20 fe00 ff30 ff00 fe02
80f8 10ff 02fe 30ff 00ff 0100 f820 fe30
ff40 30ff 0201 f830 fe20 ff20 ff00 ff00
fe30 fe22 fe40 ff42 0080 f840 ff30 ff30
ff32 fe02 81f0 20ff 00ff 22fe 02ff 0100
f820 ffb0 fe40 32fe 0000 fcb0 ff31 ffb0
fe30 fe00 10fc 00ff 30ff 30fe 4000 02fe
00ff 6030 ff20 0041 f832 ff10 fe32 fe20
ff00 00f0 b0fe b1ff 32ff 42ff 0000 f000
ff30 fe30 ff30 ff00 08f0 32ff 20fe 20ff
20fe 0000 fc31 ff00 ff20 fe31 fe02 46f8
20fe 30fe 30fe fe02 00f8 b0ff 20fe 32fe
00fe 0000 f830 fe00 fe80 fe20 ff00 00f0
32ff 40b0 fe30 ff02 fe30 fe20 fe20 ff40
ff01 41f8 30ff b0fe 20ff 30ff 0200 f8b0
ff20 ff20 feb0 fe02 01f8 30fe 20fe 00fe
00fe 0101 fe30 ff30 fe31 fe00 ff00 00fc
20fe 32fe b0fe 00ff 0100 f810 fe30 feb0
ff10 ff00 91f0 6000 ff30 ff30 ff02 01f8
20ff 30fe 30fe 30fe 0042 e0b0 feb0 fe60
00ff 0000 fcfe 32fe 32ff 31ff 0000 fe40
ff30 fe02 fe20 ff00 
```

And as soon as we start DJI sdk launch file the output content changes and then stops publishing! Here are the output after launch and after the last line nothing else is published:
```
00fc 30fe 00ff b0ff 32ff 0201 f822 fe20
ffb1 fe20 fe00 80f0 31fe ff00 ff00 fe02
aae4 0000 0000 0000 b8e5 b201 0200 7f0b
f05d 0200 0080 65e9 f05d 0200 0000 0000
0080 3f00 0000 0000 0000 0000 0000 0004
0000 0004 0000 8000 0000 8000 0000 0000
0000 006a 0000 0001 0000 0000 0000 0000
0000 0000 ca85 afab 29aa e43f e0f0 9c76
2f1b c4bf cdcc cc3d 0000 0000 05cd cccc
3d00 00f0 4100 00c8 c200 00c8 4200 00c8
4200 0000 001f c4c8 020c c0d4 0100 0000
0000 0000 0000 0000 0000 0000 0000 0000
0000 0000 0000 9a99 193f 0000 0000 0000
0000 0000 0000 0000 0000 0000 0000 0000
0000 0000 0000 0f00 0000 dc05 0000 0000
0000 0000 0000 0000 0000 0000 0006 004a
376b 6f83 
```

samsdolphin, 2018-07-27:
Have you tried using __PL2302__ usb-ttl serial cable instead of CH340? I doubt CH340-chip serial doesn't support baud rate 921600 thus you can't activate the SDK but do receive some garbage from port.

ilia-sheremet, 2018-07-27:
@samsdolphin It's not USB-to-Serial converter. On odroid XU4 there is a separate port for direct serial connection and that's the one that is failing. It worked fine when using the USB-to-Serial converter CP2102. But only on Ubuntu 16 it doesn't work with direct serial connection, on Ubuntu 14 it works.

prodrone-dev, 2018-07-27:
@samsdolphin It's not USB-to-Serial converter. On odroid XU4 there is a separate port for direct serial connection and that's the one that is failing. It worked fine when using the USB-to-Serial converter CP2102. But only on Ubuntu 16 it doesn't work with direct serial connection, on Ubuntu 14 it works.

closed 2021-06-25 by dji-kyle