dji-sdk/Onboard-SDK-ROS #177: drone activation error on m100 on branch 3.3-3.6

mithunvarma, 2018-02-12:
Initially I was on 3.3 branch, all nodes were working.
due to some reasons, I had to delete catkin workspace of Onboard-sdk-ros and after again tried to catkin_make Onboard-sdk-ros on feb 5, i cloned 3.4 for m100 as per readme and even built Onboard-sdk 3.4 too. I was getting these same errors on all branches 3.3,3.4 and 3.6[matched onbard sdk branch and sdk-ros branch]

 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [25553]
ROS_MASTER_URI=http://uas4:11311

setting /run_id to 44f897d2-1027-11e8-b391-00044b5acc6d
process[rosout-1]: started with pid [25566]
started core service [/rosout]
process[dji_sdk-2]: started with pid [25574]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS2 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.
[ INFO] [1518462149.947117806]: Succeeded to read from serial device

STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS2 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L686: Device Serial No. = 041DDA1369

STATUS/1 @ parseDroneVersionInfo, L688: Hardware = M100

STATUS/1 @ parseDroneVersionInfo, L689: Firmware = 3.1.10.0

STATUS/1 @ parseDroneVersionInfo, L692: Version CRC = 0xA6453AAC

ERRORLOG/1 @ functionalSetUp, L161: Subscriber not supported!

[ERROR] [1518462150.969182166]: drone activation error
[ERROR] [1518462150.969334403]: Vehicle initialization failed

the Flight controller version is 3.1.10.0
firmware on dji go app shows 1.3.1.0 
im using nvidia tx1 arm64

thanks

mithunvarma, 2018-02-12:
Got the same error and three tx1 devices and three m100 drones as well

rohitsantdji, 2018-02-12:
See dji-sdk/Onboard-SDK#249. A fix is on the way.

rohitsantdji, 2018-02-16:
The fix has been released. Please get the latest 3.6 branch for the osdk-core repo, and re-open if issues persist.

closed 2018-02-16 by rohitsantdji