dji-sdk/Onboard-SDK-ROS #184: Failed to read from serial device

Niboros91, 2018-03-08:
I am using Ubuntu 16.04, ROS Kinect, DJI ROS SDK 3.6 and DJI M600 Pro V1.00.01.65.

When I try to launch: $ roslaunch dji_sdk sdk.launch. I get the following error:

[image]

I have checked my serial port and cable, both are working fine with other devices, and I have also checked the baudrate (921600).
I tried as said here: dji-sdk/Onboard-SDK#176 and it didn't work 
Also doing: cat /dev/ttyUSB0 I get no data from the drone.
In dji Assistant I enable API sdk control and I am using the latest firmware.

PS: All this was working fine 2 weeks ago.

rohitsantdji, 2018-04-11:
Did you get this resolved?

asiron, 2018-05-03:
We were having the same problem. It doesn't work at all @ 921600 baud rate, but it works @ 115200, which can be insufficient if we want all the data at 100Hz. Setting manually baud rate through `stty -F /dev/ttyUSB0 921600` does not help neither.

dji-kyle, 2021-07-07:
Please check the UART module/cables support baudrate up to 1000000

closed 2021-07-07 by dji-kyle