dji-sdk/Onboard-SDK-ROS #185: M210 - Dual Main cameras and Thermal image

soutieslak, 2018-03-09:
I can successfully access the images of my main camera (Zenmuse X5s) and since the M210 has a dual mounting system for two main cameras, I want to know if it is possible to mount the Zenmuse XT (Thermal camera) on the other mount.

1. Can you have two Zenmuse x5s (two main cameras)? How would you distinguish between the two?
2. Can one of them be replaced by the Zenmuse XT (thermal camera)?
3. If I fly with only the Zenmuse XT mounted, will the thermal image come through? (Are the drivers the same for the Zenmuse x5s and XT?

Any help or comment would be appreciated.

rohitsantdji, 2018-04-11:
Currently, the OSDK can only access camera streams from the camera that is on the gimbal labeled 'I', not the one labeled 'II'. Also, for the moment, we only support streams from the X4s, X5s and the newly-launched XT2 camera. For the older XT camera, we are looking into supporting it but do not have an ETA at this time.

soutieslak, 2018-04-11:
Hi, thank you for the reply! Is there any ETA for OSDK support of camera streams from gimbal 'II'?

guicolares, 2019-02-15:
Hi, 

Also is it possible get temperature for each Pixel and change some parameters from camera with onboard SDK ? 

Thanks

alejodosr, 2019-06-04:
Hi @soutieslak I am trying to test the Zenmuse X4S (M210RTK V2) but without success, can you provide any insights about this topic? We are completely stuck, since we are running the official OSDK with supported devices and OS (Ubuntu 16.04). 

(I posted a wider explanation [this issue](https://github.com/dji-sdk/Onboard-SDK/issues/280)

dji-kyle, 2021-07-07:
Please update the firmware to the latest and OSDK 4.1(for M210 v2 and M300)

closed 2021-07-07 by dji-kyle