dji-sdk/Onboard-SDK-ROS #187: unable to run Advanced Sensing - Stereo Images sample

QE2018, 2018-03-13:
when we run rosrun dji_sdk_demo +tab we cannot find stereo images sample, but only the other six (i.e  demo_flight_control). can you tell me how can I get this one work?
thank you very much.

rohitsantdji, 2018-04-11:
Did you compile your djiosdk-core library with the -DADVANCED_SENSING=ON flag, and then perform a `sudo make install`? If you have not done that, then recompile the core library in that manner, and clean out your build and devel directories in your catkin workspace as well. If that still doesn't show you the sample, post the output of your catkin_make here.

rohitsantdji, 2018-04-25:
Closing since there has been no update for over 2 weeks. Please reopen if any questions remain.

closed 2018-04-25 by rohitsantdji