dji-sdk/Onboard-SDK-ROS #189: Virtual sticks service on 3.3

shivangbaveja, 2018-03-17:
I am using v3.3 and couldn't find the functionality for virtual sticks in the ROS sdk. Was this functionality deprecated after 3.2?

Does 3.2 work on ROS Kinetic? Which version of Onboard-SDK would be required to work with it?

amenonDJI, 2018-03-19:
@shivangbaveja Yes the Virtual stick functionality was deprecated after version 3.2. ROS Kinetic was Beta tested on 3.2. When we tested it, we did not see any obvious issues with 3.2 working with ROS Kinetic, but our test for this combination was only a light test of all the samples on the bench.

shivangbaveja, 2018-03-19:
OSDK-ROS 3.2 is working on ROS Kinetic for me. 

Only issue is, so far we have been using v3.3 which was a major release and has a lot of changes. If we were to switch to 3.2 we would have to do a lot of code changes as the topic names and their data structures are different in 3.2. 

Is there some way to use Virtual Stick API with v3.3 or v3.4?

If not, what's the best way to go about using the Virtual Stick functionality? 
Switching to 3.2 or implementing the functionality ourselves in v3.3.

amenonDJI, 2018-03-20:
@shivangbaveja It is best not to downgrade the version. You could implement this feature on 3.3 or later (we would recommend using the latest OSDK) by reading RC stick values and commanding the aircraft using one of the Control commands.

shivangbaveja, 2018-03-20:
@amenonDJI  Okay. Thanks for your feedback.

closed 2018-03-20 by shivangbaveja