dji-sdk/Onboard-SDK-ROS #190: Implementation of data transparent transmission

tngan, 2018-03-20:
In the updated documentation of data transparent transmission [here](https://developer.dji.com/onboard-sdk/documentation/appendix/mobile-onboard-APIs.html), we will need to implement the delegate in order to get invoked when there is external data from OES.

The latest ROS service for data downstream from OES to mobile is renamed from `dji_sdk/send_data_to_remote_device ` to `dji_sdk/send_data_to_mobile`. For a quick try, I executed the following command, the return is true however the delegate I implemented on mobile side isn't invoked.

```console
$ rosservice call /dji_sdk/send_data_to_mobile "data: 'testing'"
```

Then I tried to implement in code.

```c++
unsigned char data_to_mobile[100] = {0};
data_to_mobile[0] = 0x00;
data_to_mobile[1] = 0xfe;
_arbitrary_data.request.data.resize(100);
memcpy(&_arbitrary_data.request.data[0], data_to_mobile, 100);
if (dtt_service.call(_arbitrary_data))
{
  ROS_INFO("Successfully sent data");
}
else
{
  ROS_ERROR("Failed to call the service");
}
```

The return is true but it seems not invoke the delegate.

And the code of delegate is used to print a hardcoded string on a text view, in order to check if it's already invoked.

```objc
- (void)flightController:(DJIFlightController *_Nonnull)fc didReceiveDataFromOnboardSDKDevice:(NSData *_Nonnull)data {
    [self.sampleView setText:@"arbitrary_data"];
}
```

And I have also found another hint from [FAQ](https://developer.dji.com/onboard-sdk/documentation/appendix/FAQ.html).

**Unfortunately, the LightBridge 2 channel does now support transmitting arbitrary forms of data. If you have data in the form of a video stream, use N1 Video Encoder (M100) or HDMI (M600/A3/N3) to stream it back to the RC. If you have some low-bandwidth data, consider using Data Transparent Transmission.**

My whole setup includes LB2, N3, Manifold, X5. I want to know what is the limitation of using data transparent transmission (in terms of hardwares) ?

rohitsantdji, 2018-04-11:
Your hardware is compatible with the transparent transmission scenario you are mentioning. Could you 1) Tell us what iOS SDK version you are using? and 2) Try the same scenario without using ROS, by using the API directly in a Linux sample? Also, does the upstream iOS --> Onboard transmission work as expected?

rohitsantdji, 2018-04-25:
Closing since there has been no update for 2 weeks. Please reopen if any questions remain.

closed 2018-04-25 by rohitsantdji