dji-sdk/Onboard-SDK-ROS #192: M100 indoor flight problem

ChadLin9596, 2018-04-09:
Hello, we try to use optitrack to give M100 local position to accomplish the indoor flight, however, when I run this node (M100 already takeoff), M100 can't  follow my command.
Also, we have checked the command which already sent to topic "flight_control_setpoint_generic".
...
int main(int argc, char **argv){

    ros::init(argc, argv, "offb");
    ros::NodeHandle nh;
    ros::Subscriber local_position;
    ros::Publisher ctrlvelPub;
    ros::ServiceClient local_pos;

    local_position = nh.subscribe<geometry_msgs::PoseStamped>("vrpn_clinet_ros", 10, &mocap_cb);
    ctrlvelPub = nh.advertise<sensor_msgs::Joy>("dji_sdk/flight_control_setpoint_generic", 10);
    local_pos = nh.serviceClient<dji_sdk::SetLocalPosRef> ("dji_sdk/set_local_pos_ref");
    //initialize

    ros::Rate loop_rate(30);

    dji_sdk::SetLocalPosRef localPosReferenceSetter;
    local_pos.call(localPosReferenceSetter);

    while(ros::ok()){
      // input goal;
      keyboard_control();
      // local_position are obtained by optitrack
      geometry_msgs::PoseStamped err;
      err.pose.position.x = goal.pose.position.x - local_position.pose.position.x;
      err.pose.position.y = goal.pose.position.y - local_position.pose.position.y;
      err.pose.position.z = goal.pose.position.z - local_position.pose.position.z;
      ROS_INFO("command : %.4f, %.4f, %.4f",err.pose.position.x,err.pose.position.y,err.pose.position.z);
      
      double  xCmd ;
      double  yCmd ;
      double  zCmd ;
      double  yawDesiredRad= 0  ;

      xCmd = err.pose.position.x;
      yCmd = err.pose.position.y;
      zCmd = err.pose.position.z;

      sensor_msgs::Joy controldata;
      controldata.axes.push_back(xCmd);
      controldata.axes.push_back(yCmd);
      controldata.axes.push_back(zCmd);
      controldata.axes.push_back(KP_YAW*yawDesiredRad);
      controldata.axes.push_back(flag);
      ctrlvelPub.publish(controldata);

      ros::spinOnce();
      loop_rate.sleep();
    }
    ROS_INFO("Flight Control Ending");
}
...

rohitsantdji, 2018-04-11:
Did you obtain control authority by calling the control authority service?

ChadLin9596, 2018-04-11:
Yes I have done this and the obtain_control_result is true.

reygago, 2018-04-11:
Your flag seems to be wrong. You can command with DJISDK::HORIZONTAL_ANGLE (set roll and pitch) or with DJISDK::HORIZONTAL_BODY (set x and y), not both.
From your code, it seems to me that you set  the local reference using optitrack information and then set a position command, so I would use only DJISDK::HORIZONTAL_BODY flag. 
I'm not sure if this will work because I've never used the set local reference service, usually I save all the positions references in an external computer and just command using roll and picth increments until I read the right position.

rohitsantdji, 2018-04-11:
Upon inspecting your code more closely, I see a few issues:

1. The `SetLocalPosRef` service sets a local ENU origin at your current GPS location. If you are indoors, this will not work since GPS is not available.

2. As @reygago says, your flag is trying to command attitude and you are sending it position values. You need to convert the position errors to attitude commands.

3. You will need to also be sure that the magnetometer is reporting no errors, since if it is reporting errors the angle control will fail.

4. Finally, your vertical THRUST flag is not appropriate for sending position commands, and also quite dangerous.

ChadLin9596, 2018-04-13:
Hi @reygago thanks your advice, 
 `SetLocalPosRef` in this code is not be used at all, local_position are got from `vrpn_clinet_ros` , also flag of `VERTICAL_POSTITION` might not be used without GPS, so my strategy is use P control to let convert position input into thrust command.

Hi @rohitsantdji thanks your advice,
How to check the magnetometer no error, also I want to ask that what frame is when I used flag of  `HORIZONTAL_ANGLE` . I simulated M100 flight in DJI assistant 2 and found that frame is not like this.
[image]

ChadLin9596, 2018-04-18:
hi @rohitsantdji thanks your all advices.
I solved these problem by these step.
1. run the node first 
2. use remote controller to takeoff, then M100 received my command.

closed 2018-04-18 by ChadLin9596