dji-sdk/Onboard-SDK-ROS #196: Unable to compile the packages in an OnBoard Computer

sarathjayanthan, 2018-04-24:
Hello,

I am trying to compile these packages in a Raspberry Pi  (Raspbian Stretch intalled in it). By following instructions from the below link, I created a workspace and installed ROS Kinetic. 

http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

Also, by following the instructions in the below link, I have installed the necessary packages for On-Board SDK.
http://wiki.ros.org/dji_sdk/Tutorials/Getting%20Started

On compiling I am getting this error:

[image]

Below is a screenshot of the packages inside my workspace/src, just for a reference:
[image]

Could you help me solve this issue, please?

Thanks.

rohitsantdji, 2018-04-25:
DJI OSDK ROS assumes (implicitly, through the dependencies set up in package.xml) that you have the image_transport package installed - it appears you don't. So you will need to install it.

On a side note, are you able to compile any ROS packages, apart from DJI SDK?

sarathjayanthan, 2018-05-02:
Thanks @rohitsantdji for the reply. 
I did not compile any package separately. So, can't say whether any ROS Package got compiled completely.

Infact, I have now installed Ubuntu MATE 16 with ROS Kinetic in the Raspberry Pi. Now trying to install Onboard DJI-SDK Packages.

Right now, I am facing the issue that I cannot compile the packages completely (only till 96%). Also, tried catkin_make with different cores (-j1, -j2, -j3), with no luck. The compiling gets stuck at around 96% and the screen becomes frozen. I also tried catkin_make _isolated also.

Could you help, please?

sarathjayanthan, 2018-05-02:
I connected to Raspberry Pi through SSH from my PC and compiled the packages (catkin_make -j1). This worked.

closed 2018-05-02 by sarathjayanthan