dji-sdk/Onboard-SDK-ROS #198: ROS Stereo Vision Depth sample will not be compiled

imkhairulikhwan, 2018-04-26:
Hi and good day guys, 

I open up new issue as I think it's different from the previous one I opened. I hope it's not a duplicate one. I uninstall and re-install again using the steps mentioned below:

https://docs.opencv.org/trunk/d7/d9f/tutorial_linux_install.html

and the installation seems to be fine. However, when I tried to run cmake, I still got "Did not find required libraries, stereo vision depth perception sample will not be compiled." error message, without mentioning any specific libraries, as per attachment. Previously I got similar message but it did mentioned few missing libraries in the message. 

Thank you and have a nice day! 

[image]

imkhairulikhwan, 2018-04-26:
After I re-installed opencv, I am able to get the sample compiled. But unfortunately, it's a new error now. When I tried to catkin_make, I got error "const struct darknet_ros_msgs::BoundingBoxes_<std::allocator<void> >’ has no member named ‘boundingBoxes’"
, as per attachment. 

[image]

imkhairulikhwan, 2018-10-18:
Hi @chenzheng2016 , sorry for the really late reply. Not sure if you already solved it, but I will just comment it out here as reference to the others. 

I noticed that ROS comes with their own OpenCV package, what I did was not to install ROS full desktop. I installed ROS using bare packages, then install the other ROS packages individually. 

during catkin_make, you can specify you own OpenCV path, that is if you installed it manually from the source. In my case:

catkin_make -DOpenCV_DIR=/usr/ocal/share/OpenCV

Hope it helps!

langelesl, 2019-03-01:
Hello, @imkhairulikhwan Today I tried to compile another package in ROS and met the similar error about lack of members "boundingBoxes". Have you solved this problem or find the cause? Thanks for help.
[image]

lanyusea, 2019-03-02:
@langelesl  he already said it: https://github.com/dji-sdk/Onboard-SDK-ROS/issues/198#issuecomment-430882747

langelesl, 2019-03-02:
I thought the posted comment you've mentioned is only related to the first CMake error (about opencv) he met. Did it also solve the error of "can't find member 'boundingBoxes' in darknet_ros_msgs"? I've checked the BoundingBoxes.msg file in darknet_ros_msgs, there is the only member named "bounding_boxes". I just tried this name instead of "boundingBoxes" in .cpp and compiled successfully. It seems to be solved temporarily. Thank you for your reply @lanyusea