dji-sdk/Onboard-SDK-ROS #199: ROS "called vehicle->control->goHome()"

a137755952, 2018-05-02:
takeoff_land(dji_sdk::DroneTaskControl::Request::TASK_GOHOME);

When I called this service the drone is landing without going to the original position.

Can anyone solve the problem?

rohitsantdji, 2018-05-02:
Is the drone within a 10m radius of the original takeoff point? If it is, then this is expected behavior.

a137755952, 2018-05-03:
@rohitsantdji  I set only one point by setTarget(15, 5, 10, 60)
I think this is more than 10m radius. But If I setTarget(30,5,10,60), the drone will hover at（20，5，7，60），and the xCmd、yCmd and zCmd is 2、0 and 11. So I do not understand why the drone is finally stop at(20,5,7,60）

rohitsantdji, 2018-05-16:
The SDK is almost entirely REP-103 compliant, so that means when you want to command position, the values are in meters, when you are commanding yaw, the value is in radians.

The issue you described sounds like a flight restriction enforced by the drone. Open the DJI Go app, make sure you are connected to the internet at least once, and turn off any Beginner Mode switch or any message you are seeing from the app. If you are in China, you will have to log in to the app to unlock full functionality. If you are not in China, make sure your simulated co-ordinates are not located in China.

closed 2021-07-07 by dji-kyle