dji-sdk/Onboard-SDK-ROS #200: ERRORLOG/1 Connect to MAIN_CAMERA failed, Error: Connection setup failure: connection time out.

a137755952, 2018-05-04:
when I try this cmd : ./camera-stream-target-tracking-sample UserConfig.txt , the terminal show that:

STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ initGimbal, L850: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ initGimbal, L894: Gimbal not mounted!

ERRORLOG/1 Connect to MAIN_CAMERA failed, Error: Connection setup failure: connection time out.

ERRORLOG/1 Initialize MAIN_CAMERA failed
Double check USB connection or re-plug in USB cable.
Failed to Open Camera

Is there anyone can help me?

rohitsantdji, 2018-05-16:
Can you tell us the following information?

1/ What camera/gimbal is connected to the aircraft? 
2/ Do you have a single-gimbal or dual-gimbal M210? If dual, which side is the camera mounted on?
3/ Is the remote controller ON and connected during this operation?

zhaxiche, 2018-11-12:
> Can you tell us the following information?
> 
> 1/ What camera/gimbal is connected to the aircraft?
> 2/ Do you have a single-gimbal or dual-gimbal M210? If dual, which side is the camera mounted on?
> 3/ Is the remote controller ON and connected during this operation?

Hi, I have the same issue. The camera is ZENMUSE X4S, it is a dual-gimbal M210 and I mounted it on the right hand side if facing the front of the drone, and the remote controller is ON and connected, but I didn't open Clearsky screen.

jonatanolofsson, 2020-10-27:
I have this issue as well - camera on the right side, remote connected, ZENMUSE X4S on a M210 v1. Really a shame DJI dropped support for the v1 so fast in the SDK, but did this issue ever get a solution? Trying with SDK 3.9, and everything except the camera seem to be working.

Output from `./camera-stream-callback-sample UserConfig.txt /dev/ttyACM0`: 
```
STATUS/1 @ init, L56: Attempting to open device /dev/ttyACM0 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

ERRORLOG/1 Connect to MAIN_CAMERA failed, Error: Connection setup failure: connection time out.

ERRORLOG/1 Initialize MAIN_CAMERA failed
Double check USB connection or re-plug in USB cable.
Failed to open selected camera
```

dji-kyle, 2021-07-07:
Please update the formware to the latest and OSDK version 3.0 for M210 v1. And test it with RC app connected.

closed 2021-07-07 by dji-kyle