dji-sdk/Onboard-SDK-ROS #202: Failed when make djiosdk-core, undefined reference to cuda

imkhairulikhwan, 2018-05-05:
Hi and good day,

previously I was able to make djiosdk-core but that was before installing cuda (after long time figuring way to do that) with opencv. But afterwards, I got few errors. 

At first, I received two errors of not finding files:

1. opencv2/cudastereo.hpp
2. opencv2/cudawarping.hpp

as stereo_frame.hpp is referring to /usr/local/include/opencv2, what I did was just copying those two files from opencv source folder into that folder.

Not sure whether it's the correct way to fix it, but there was no longer that error message. 

But then, stereo_frame.cpp throws another error; undefined reference to cv::cuda::createStereo....
as per attachment.

I guess I am still missing some steps perhaps?

Thank you in advance! 

[image]

rohitsantdji, 2018-05-16:
Hi @imkhairulikhwan , your CUDA version seems to be 9.1. On our [samples](https://developer.dji.com/onboard-sdk/documentation/sample-doc/advanced-sensing-stereo-depth-perception.html) page, we recommend using OpenCV 3.3.1, OpenCV Contrib 3.3.1, and CUDA 9.0 - we have not tested with different combinations of CUDA/OpenCV.

imkhairulikhwan, 2018-05-16:
Hi @rohitsantdji , noted on that, I will proceed to re-installing the packages to match with the version. I will then revert back the result. Thank you!

imkhairulikhwan, 2018-05-23:
I also tried to copy the respective files into the opencv2 include folder, which are cudastereo.hpp and cudawarping.hpp but still got the following error: 

[image]

imkhairulikhwan, 2018-05-28:
Hi guys, any updates on this?

thanks.

rcabg, 2018-06-22:
Hi @imkhairulikhwan ,
Did you compile opencv with WITH CUDA?

Also, do you have ROS-desktop-full installed? If so, ROS-desktop-full comes with a non Cuda OpenCV. Please check if cmake is linking to the correct version of OpenCV with CUDA. I hope that helps!

zliDJI, 2018-06-25:
@imkhairulikhwan  Copying header files from source to installation folder was not the right way to solve the problem.

Did you follow the exact steps in the Appendix of this page? https://developer.dji.com/onboard-sdk/documentation/sample-doc/advanced-sensing-stereo-depth-perception.html

imkhairulikhwan, 2018-07-04:
Hi @rcabg , @zliDJI , yes we have compiled opencv with CUDA, when trying to cmake opencv with CUDA, it seems to be installing just fine, as per screenshot. 

Yes, we did follow the exact steps as mentioned in the Appendix page as you mentioned. (with the exact same versions of OpenCV, Distrib, and CUDA)

[image]

imkhairulikhwan, 2018-07-04:
Another attempt, it failed at 100%

[image]

zliDJI, 2018-07-05:
Hi, do you have ROS installed? Could you check if there is a folder /opt/ros/kinetic/include/opencv-3.3.1-dev  ?
Or you can remove everything under build folder and re-run:
cmake .. -DADVANCED_SENSING=ON again and post your cmake output.
Thanks!

imkhairulikhwan, 2018-07-05:
Hi, the new machine I tried does not have ROS installed. May I know if it's necessary to have ROS to run the sample with -DADVANCED_SENSING=ON? Although I have tried with other machine, but if ROS is necessary then I will proceed to install ROS first.

Thank you!

zliDJI, 2018-07-05:
Quite opposite. I'm suspecting that ROS comes with another opencv 3.3.1 but with different build configuration (without cuda support), and the cmake picked that version instead of the one you built and installed manually.

Could you clean up your build folder and run cmake again and let us know the result?

imkhairulikhwan, 2018-07-05:
I see. I guess it should be fine then without ROS installed. 

But yes, I will clean up the folder and let you guys know on the result.

Thanks!

imkhairulikhwan, 2018-07-06:
@zliDJI my mistake, as we were installing in few machines, was not aware that the current one does have ROS installed. 

After checking, we do have /opt/ros/kinetic/include/opencv-3.3.1-dev folder. How do I check if cmake is linking to the correct opencv with CUDA support? 

Thank you!

rcabg, 2018-07-06:
@imkhairulikhwan hey,

you can print with a MESSAGE(...) the opencv variables that your project links in the following way:
`MESSAGE("OpenCV = ${OpenCV_VERSION} - ${OpenCV_LIBS}")`

`${OpenCV_LIBS}` will show the used OpenCV library path.

As I posted before, ROS-desktop-full comes with a non Cuda OpenCV. What I did was reinstall ROS without its default OpenCV. Then I compiled my own OpenCV version with CUDA and finally, I compiled from source the ROS OpenCV dependent packages, such as CV_BRIDGE. That way you will only have one OpenCV version to link with.

zliDJI, 2018-07-06:
Hi, @rcabg , thank you so much for taking the effort to solve the conflict between the opencv with ros and opencv built/installed manually!  

The setup.bash script in ros set some environmental variables which make the cmake find ros's opencv first.  In order to prevent this behavior, a quick solution is to **not run** 

> source /opt/ros/kinetic/setup.bash

@imkhairulikhwan  Please check you ~/.bashrc to see if you have sourced the setup.bash by default. If so, please comment out those lines (by putting # sign before the line), and save the .bashrc file.  Then **start a new terminal**, and run the cmake process again.

Please post the cmake result to see if cmake is using the opencv from /usr/local or still from /opt/ros/kinetic . Thanks!

imkhairulikhwan, 2018-08-04:
@zliDJI Sorry for the late reply, I was able to run the cmake successfully now, but failed when running catkin_make, which I guess it's due to that it detects the opencv in ROS include folder instead of the actual I built initially. By commenting the setup.bash, it would not allow me to run the catkin_init_workspace so I need to reinstall ROS again without its default OpenCV right? As suggested by @rcabg. But may I know how do I do that? Sorry because I am not really familiar with the build in Linux (was using Windows all this while :/ )

Thank you in advance!

closed 2021-07-07 by dji-kyle