dji-sdk/Onboard-SDK-ROS #206: Unable to rostopic echo subscriber topics

sarathjayanthan, 2018-05-15:
Hello,

I am running Onboard DJI-SDK Packages in a Raspberry Pi 3 (Ubuntu Mate 16 and ROS Kinetic) connected to A3 Microcontroller (Using DJI Assistant 2 Simulator). I am able to run the following commands successfully:

**roslaunch dji_sdk sdk.launch
rosrun dji_sdk_demo demo_flight_control**

But when I try 'rostopic echo' of any of the following topics, nothing is displayed:
_/dji_sdk/flight_control_setpoint_ENUposition_yaw (sensor_msgs/Joy)
/dji_sdk/flight_control_setpoint_ENUvelocity_yawrate (sensor_msgs/Joy)
/dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition (sensor_msgs/Joy)_

Could someone tell me how i can visualize these subscribed values? (even rqt_graph is not working in Ubuntu Mate)?

Thanks

rocklinsuv, 2018-05-16:
The topics you listed are the topics where control commands are published to. Of course it doesn't show anything if you are not sending control commands! Try /dji_sdk/imu or /dji_sdk/battery_state instead.

sarathjayanthan, 2018-05-16:
Thanks for the reply @rocklinsuv.

Does that mean when these topics are subscribed to , the remote control commands are overridden? 

Also, rqt_graph is also not working properly. Is there any way to visualise the ROS Computation Graph? That would be very helpful in understanding how the communication is happening.

rocklinsuv, 2018-05-16:
@sarathjayanthan 

> Does that mean when these topics are subscribed to , the remote control commands are overridden?

I don't quite get it. What do you mean by "the commands are overriden"? 
Also you can't subscribe to these specific topics, but instead you need to publish messages to them.

> Also, rqt_graph is also not working properly. Is there any way to visualise the ROS Computation Graph?
"rosbag record /topic" combined with "rqt_bag name_of_bag_file" would be a feasible way to visualize the topics.

sarathjayanthan, 2018-05-16:
When these topics are published to, will the commands from the remote control work? During 'rosrun dji_sdk_demo demo_flight_control', remote control commands were disabled.

rocklinsuv, 2018-05-16:
When you say "remote control command", did you mean the R/C transmitter?

> When these topics are published to, will the commands from the remote control work?

If you are referring to the R/C transmitter, during "demo_flight_control", none of the R/C command execept the one which turns the props off (which I strongly recommend you to try in the simulator only) will work.

> But during that, none of these topics are published to. Then which are the topics that take remote controller input and pass it to the drone?

Do you mean taking back control if things go south? You can always flick the control mode switch on your R/C back to "A" or "P" mode, then the control commands from "demo_flight_control" won't take effect.
Alternatively, I suppose you may subscribe to "dji_sdk/rc", and manipulate the RC output as input and feed it back to your drone via your own control algorithm. I haven't done that myself, but I suggest you Google it. Maybe someone else has done it before.

rohitsantdji, 2018-05-16:
Thanks for your helpful input @rocklinsuv! 

@sarathjayanthan to add some more information: The remote controller does not control the drone through the SDK. You can override the remote controller's commands by publishing commands to the topics you mentioned. The [Control Authority Guide](https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-control-authority.html) will be helpful to understand the flow and priority further.

sarathjayanthan, 2018-05-29:
Thanks a lot! @rocklinsuv and @rohitsantdji

closed 2018-05-29 by sarathjayanthan