dji-sdk/Onboard-SDK-ROS #209: DJI Go with DJI Assistant 2 Simulator

sarathjayanthan, 2018-06-05:
Hello,

 Is it possible to use DJI Go (in iOS or Android) and command the drone in DJI Assistant 2 Simulator? I am using DJI M600 Pro with On-Board Computer (with DJI SDK ROS Packages).

Thanks for any help.

rohitsantdji, 2018-06-05:
Yes. Do note that the DJI Go control with Assistant 2 simulation is completely independent of Onboard SDK.

sarathjayanthan, 2018-06-05:
Thank you for the reply. 

Actually, I would like to enable/disable 'Multiple Flight Mode' option for real-time flight (without simulator). I understand that it can be done through DJI Go. Is it possible to do the same with Onboard SDK?

closed 2018-06-12 by sarathjayanthan

rohitsantdji, 2018-06-12:
You can monitor the mode switch state by subscribing to RC topics, or using the RC telemetry Broadcast data.